MarlinFirmware/Marlin #11340: BLTOUCH ISSUE ON Y AXIS

liljuanton, 2018-07-23:
Machine Info:
FT-5 R2 (X0,Y0 is at the top rear right )
Authentic BLTOUCH V2 (with the new clear/plastic probing pin.)
TMC 2208 (Connected to software TX/RX on AUX 2) 
Tita Aero Hotend 
PEI covered borosilicate printing surface.
Current 1.1 BUGFIX installed (although issue was also present on 1.1BUGFIX from a month or so ago)
MKS GEN 1.4
-->

BLTOUCH Unable to probe beyond 153 in the Y axis, No issue anywhere on the X axis . M48 Fails immediately without attempting to probe and sends the bed 10MM in the opposite direction, Finishes with this error code

Send: M48 P4 V4
Recv: M48 Z-Probe Repeatability Test
Recv: Positioning the probe...
[...]
Recv: Error:STOP called because of BLTouch error - restart with M999
Recv: Error:STOP called because of BLTouch error - restart with M999
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 143.000 Y: 170.000 Z: 0.000
Recv: Error:Probing failed
Recv: X:92.00 Y:170.00 Z:25.47 E:0.00 Count X:7360 Y:13600 Z:10188
Recv: ok
[...]

Currently Config.H is setup to probe 4 points in the X and Y Axis. Error happens when using ABL Bilinear on 9th probe position, prior 8 positions that were probe complete without issue. Issue persist wether unit is set to probe: 9 or 16 positions, fast or slow, once or multiple probes. I've also setup the BLTOUCH on both the  Z_MAX_PIN and Z_MIN_PIN (currently connected to Z_MIN_PIN) Issue is present regardless of Pin used. Additionally BLTOUCH is able to complete M48 V4 P10 with a standard deviation of less than .0025 anywhere on the X axis and anywhere unto 153 in the Y axis, however fails instantly if probed at 154 or beyond

### Steps to Reproduce

enabled AUTO BED LEVELING BILINEAR, configured to probe once, 4 points per axis (16 total pts), have attempted enabling and disabling BLTOUCH Delay up to 500ms, MIN_PROBE_EDGE 10mm, attempt to run G29  and any probe points beyond 154mm on Y axes (unit appears to miss fire) and fails immediately. also unable to run M48 V4 P4 @ 154mm and beyond unit failed with error bellow 

Send: M48 V4 P4
Recv: M48 Z-Probe Repeatability Test
Recv: Positioning the probe...
Recv: Error:STOP called because of BLTouch error - restart with M999
Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: Bed X: 272.000 Y: 154.000 Z: 0.000
Recv: Error:Probing failed
Recv: X:221.00 Y:154.00 Z:18.01 E:0.00 Count X:17680 Y:12320 Z:7202
Recv: ok

**Expected behavior:** 
BLtouch should be able to complete G29 without error. should be able to run M48 anywhere within the defined boundary.

**Actual behavior:** 

Unit failes on 9th probing position of 16, without attempting to probe (see video). Unit is also unable to complete M48 commands @ 154 on the Y axis or Beyond, no issue anywhere on the X axis when attempting M48 commands. 

#### Additional Information

I've attempted enabling and disabling the following features:
#define MIN_PROBE_EDGE 
#define MULTIPLE_PROBING
#define MIN_SOFTWARE_ENDSTOPS
#define RESTORE_LEVELING_AFTER_G28
#define ENABLE_LEVELING_FADE_HEIGHT
#define SEGMENT_LEVELED_MOVES
#define G26_MESH_VALIDATION
#define LEFT_PROBE_BED_POSITION 
#define RIGHT_PROBE_BED_POSITION 
#define FRONT_PROBE_BED_POSITION 
#define BACK_PROBE_BED_POSITION 
#define EXTRAPOLATE_BEYOND_GRID
#define ABL_BILINEAR_SUBDIVISION 
#define BED_CENTER_AT
#define SERVO_DELAY

ALL are currently enabled 

(Video Of Issue when running G29)
https://www.youtube.com/watch?v=KXzOn4ikOVk

MY CONFIG.h, CONFIG_ADV, and pins_RAMPS as this is the only thing I've made modifications to 

[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/2217565/Configuration.h.zip)
[pins_RAMPS.h.zip](https://github.com/MarlinFirmware/Marlin/files/2217566/pins_RAMPS.h.zip)
[Configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/2217567/Configuration_adv.h.zip)

Side note I have my Filament Runout Sensor connected to my Y_MAX_PIN, I've disconnected the sensor and the issue persist

I also have a M502 followed by M500 as part of my post firmware update Gcode as I use octoprints FRIMWARE UPDATER plugin utility to wirelessly update firmware. 

I've posted to the re rap forum and Folger Forum all without a single response.

I'm not sure if theres been any changes with the new BLTOUCH  SMART V.2 that I'm running which is causing the issue. 

Important note BLTOUCH is able to complete deploy, stow and self test anywhere on the bed including @ 154mm and beyond on the Y axis, so issue does not appear to be  a physical connection/ electrical issue and appears to be a firmware based issue as Bltouch only fails when running bed leveling/ or M48 commands at 154mm or beyond on the y axis. 

I'll Happily test any recommendations and provide results 

I've  spent the last week troubleshooting this and have attempted a prior version of bug fix 1.1 and the current version as of yesterday.

liljuanton, 2018-07-23:
Issue has been isolated to a cable issue. Although the BLTOUCH was able to probe, stow and self test all without issue anywhere on the Y Axis. I found that if I moved the y axis where the probing and M48 issue was present and ran a M119 the z end stop showed as triggered and if I moved my cable chain in a certain position I could run as many M119 and it would display as open, then if I moved it back and tried it went back to a triggered state when sending a M119. 

So after everything I solved it my self, It only cost me almost a week and possibly my marriage and time with kids,

closed 2018-07-23 by liljuanton

Suterusu1337, 2019-03-08:
Stranger on the Internet, you saved me MUCH headache.  I was using an extension cable and it was lose.  My BL Touch would error out every time on the 3rd point.

After snugging the cable up, I could run G29 without issue.

Praise.