MarlinFirmware/Marlin #11341: Enabling SD power recovery

mogey336, 2018-07-23:
How can i enable SD power recovery?Do i need a hardware (power detection circuit)?

cjsoong, 2018-07-23:
1.
 Configuration_adv.h
  #define POWER_LOSS_RECOVERY..........Enable
  #if ENABLED(POWER_LOSS_RECOVERY)
    //#define POWER_LOSS_PIN   44     // Pin to detect power loss
    //#define POWER_LOSS_STATE HIGH   // State of pin indicating power loss
  #3851 

2. POWER_LOSS_RECOVERY.H
#define SAVE_EACH_CMD_MODE.......Enable

mogey336, 2018-07-23:
I have a other question
I have version 1.1.x i didn't find the power_loss in Configuration_adv.h
I have to install bugfix-2.0.x , i just copy what inside bugfix folder to my original firmware(1.1.x) or what?

cjsoong, 2018-07-24:
@mogey336  
  You need to download the new bugfix 1.1x to replace your old firmware and then set your machine parameters.
    Hardware is detecting AC power failure, if you don't have this feature, you don't need it.

takno91, 2018-07-26:
Marlin-bugfix-1.1.x   it Marlin-bugfix-1.1.x shifted axis z

mogey336, 2018-07-26:
@takno91 same problem

cjsoong, 2018-07-26:
@mogey336  @takno91  Modify this
power_loss_recovery.cpp
1. Line 141:   // strcpy_P(job_recovery_commands[ind++], PSTR("G1 Z2"));     
2. Line 161:  dtostrf(job_recovery_info.current_position[Z_AXIS] + 2, 1, 3, str_1);
                   -->  dtostrf(job_recovery_info.current_position[Z_AXIS] , 1, 3, str_1);

takno91, 2018-07-26:
cjsoong
Thanks very very very much
You helped me and saved him a great effort. I wish you happiness in this life

closed 2018-07-27 by thinkyhead

mogey336, 2018-07-27:
@cjsoong 
I edited the
line 141   strcpy_P(job_recovery_commands[ind++], PSTR("G1 Z0"));  
line 161 dtostrf(job_recovery_info.current_position[Z_AXIS] , 1, 3, str_1)
Same problem shifted axis z value 2mm 
Any ideas?

cjsoong, 2018-07-28:
@mogey336  Try this change, my test is OK
ultralcd.cpp
Line 880:
        #elif !IS_KINEMATIC
          "  X Y"
----->
        #elif !IS_KINEMATIC
          "  R0 X Y"

mogey336, 2018-08-03:
@cjsoong 
Thanks man it works now

fovarart, 2018-08-05:
i have tried all of it but some times it moves 1.5mm and some times 6mm any idea..?

jhonf441, 2018-08-17:
to use the function, it is necessary to activate EEPROM?

AnHardt, 2018-08-17:
> to use the function, it is necessary to activate EEPROM?

Yes. That's where the state data can survive.

GeekAbdou, 2018-08-18:
I've The Same Problem .. The First Resume Survive But Any Resume After It Z axis Shifted 2 , 4, 6 mm each time @cjsoong  Thank You For Contributing ^^

gonzalo11, 2018-08-25:
same problem any solution ?

gonzalo11, 2018-08-25:
the first print resume is fine but in the second it start from a correct z level but make only one layer and then level up to high and print

thinkyhead, 2018-08-25:
Power Loss Recovery is still very new and needs more refinement. For example, you cannot resume a file in a sub-folder right now, and it uses much more SRAM than necessary. It will be getting an overhaul very soon to address these issues, and we'll probably figure out the failed second resume issue also.

walidsmart, 2018-10-31:
same problem any solution ?

thinkyhead, 2018-10-31:
I'll be working on the feature later this week.

walidsmart, 2018-11-01:
i waiting you thanks so much

apaltd, 2018-11-15:
i checked for the problem, the current position for Z axis is not written in the resume file, so after resume, firmware can't define Z axis position and it blinks question mark.

thinkyhead, 2018-11-15:
### `power_loss_recovery.cpp` lines 252-253:
```cpp
    // Machine state
    COPY(job_recovery_info.current_position, current_position);
```
### `power_loss_recovery.cpp` lines 185-192:
```cpp
        dtostrf(job_recovery_info.current_position[Z_AXIS] + 2, 1, 3, str_1);
        dtostrf(job_recovery_info.current_position[E_AXIS]
          #if ENABLED(SAVE_EACH_CMD_MODE)
            - 5
          #endif
          , 1, 3, str_2
        );
        sprintf_P(job_recovery_commands[ind++], PSTR("G92.0 Z%s E%s"), str_1, str_2); // Current Z + 2 and E
```
If the position is blinking, this is because the `axis_known_position` and `axis_homed` bits are not being set to 0x07.

thinkyhead, 2018-11-16:
See #12440 for an updated and optimized version of Power-Loss Recovery. Needs testing.

apaltd, 2018-11-17:
thnaks a lot

sparky1510, 2019-05-01:
Where does it write the saved information to is it the sd card or eeprom
I'm assuming it must be the eeprom as it works if your printing directly from pc or pi or whatever you use or do you need a blank sd card in to save the file 
How's it work

blueeagle69, 2019-07-26:
As far as I am aware recovery data is saved on the sd card

hechen2019, 2019-08-11:
hello，Are there any updates about powerlosss recovery ？
I used marlin1.1.9，I  has the same problem 。First resume no problem，but the second time ，the Z AXIS postion is wrong，I see， the second time ，IF powerloss trigger at  Z height 5mm，resume print，the first layer is right，but the second layer，Z Move up about 5mm，LCD display Z 5mm（The actual distance from the platform is about 10 mm.）， the actual Z position does not match the coordinates displayed by the LCD at second layer，So nozzle hanging printing。

ShriganeshBorse, 2019-08-13:
I have marlin 2.0.x latest, I have enabled POWER_LOSS_RECOVERY and have MKS DET Power outage detection module connected to P44 , it is HIGH with we have POWER and LOW when there is no Power so I have changed POWER_LOSS_STATE to LOW.
On power outage LCD is is up for about 2 to 3 sec so I assume that is enough time to store state on SD card
But this does not seem to work 
I have ensure that power outage module works by checking with M43 P44

Without power
SENDING:M43 P44
PIN:  44   Port: L5        E_MUX2_PIN                             Input  = 0    TIMER5C   PWM:     0    WGM: 1    COM5C: 0    CS: 3    TCCR5A: 1    TCCR5B: 3    TIMSK5: 0
.                          E1_CS_PIN                              Input  = 0
>>> M43 P44

With Power
SENDING:M43 P44
PIN:  44   Port: L5        E_MUX2_PIN                             Input  = 1    TIMER5C   PWM:     0    WGM: 1    COM5C: 0    CS: 3    TCCR5A: 1    TCCR5B: 3    TIMSK5: 0
.                          E1_CS_PIN                              Input  = 0

I am not sure what I am missing here

apaltd, 2020-04-12:
Hi, upgraded to Marlin bugfix2, un-commented power_loss_recovery. not using any pin for power outage and after each layer marlin writes in PLR file on Sd card but after power outage and re plug, no resume print appears? any help will be appreciated.

thinkyhead, 2020-04-12:
Download `bugfix-2.0.x` and see if the patch just applied fixes things. It worked for me.

apaltd, 2020-04-13:
thanks for reply, downloaded version was 2.0.5.3 , i just un commented power_loss_recovery, problem exists. did i miss anything to do?

TheMiguelBi, 2020-05-01:
> Scarica `bugfix-2.0.x`e vedi se la patch appena applicata risolve le cose. Ha funzionato per me.

Hi thinkyhead,
When the power returns, even with the latest version, no pop-up window appears for recovery.
I noticed, however, that if you start a new print from the same SD you used to print with before, the superimposed window asking to "resume or delete" appears.
Printing resumes regularly (I have other problems with Z offset that I am trying to solve, but printing resumes).

_(I don't know how it worked in previous versions. I am new to this)_

TheMiguelBi, 2020-05-01:
> Download `bugfix-2.0.x` and see if the patch just applied fixes things. It worked for me.

My post above is not entirely correct. Today I was able to deepen with other tests.

- **Everything works regularly using an LCD12864 display** or using a TFT35 display simulating an LCD display.
The print recovery window appears, printing can be resumed, the coordinates are regular, etc. etc.

- The problem is using the **BTT_TFT35_E3 display** in graphic mode (via RS232). **Nothing works here!**

Here I have tried to explain the problem in detail:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/644#issue-610793577

I think the problems are in the BTT display firmware and not in Marlin.