MarlinFirmware/Marlin #11342: Bugfix-2.0x: TMC2208 Noisy and Microstepping not changeable with soft-serial

NSchnitzler, 2018-07-23:
Hi,

i have upgraded to the actual Bugfix-2.0x branch this morning and adopted the old settings. The result was really noisy stepper motors and the micro stepping setting of the configure_adv.h were not overtaken. The motor current was deployed to the stepper drivers by soft serial. The noisy stopped after enabling the stealth chop mode.

The Communication pins of the Drivers were connected to:
X 63
Y 40
Z 42
E0 65

The configure_adv.h looks like:
```#define HAVE_TMC2208
#if ENABLED(HAVE_TMC2208)  // Choose your axes here. This is mandatory!
  #define X_IS_TMC2208
  //#define X2_IS_TMC2208
  #define Y_IS_TMC2208
  //#define Y2_IS_TMC2208
  #define Z_IS_TMC2208
  //#define Z2_IS_TMC2208
  #define E0_IS_TMC2208
  //#define E1_IS_TMC2208
  //#define E2_IS_TMC2208
  //#define E3_IS_TMC2208
  //#define E4_IS_TMC2208
#endif

#if ENABLED(HAVE_TMC2130) || ENABLED(HAVE_TMC2208)
  #define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       false  // Interpolate X/Y/Z_MICROSTEPS to 256
  #define X_CURRENT          1200  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256
  #define Y_CURRENT          360
  #define Y_MICROSTEPS        16
  #define Z_CURRENT          700
  #define Z_MICROSTEPS        16
```

M122 returns:
```
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	1200	360	700	430
RMS current	1160	607	1215	773
MAX current	1636	856	1713	1090
Run current	20/31	10/31	21/31	13/31
Hold current	10/31	5/31	10/31	6/31
CS actual		0/31	0/31	0/31	0/31
PWM scale		0	0	0	0
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
tstep		0	0	0	0
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time		0	0	0	0
blank time	16	16	16	16
hysteresis
-end		-3	-3	-3	-3
-start		1	1	1	1
Stallguard thrs
DRVSTATUS	X	Y	Z	E0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:	X = 0x00:00:00:00
Y = 0x00:00:00:00
Z = 0x00:00:00:00
E0 = 0x00:00:00:00
```

ikarisan, 2018-08-03:
Hi!

Did you copy your old config?
An hour ago switched from 1.1.x-bugfix  to 2.0.x-bugfix and merged by old configuration by using a diff tool. I noticed that there is no "#if ENABLED(HAVE_TMC2208)" any longer. Things I would do are:

- M503
==> Write down all necessary values (Z offset, current, M92 values....)
- Use the new version of the configuration files and adopt the old settings as needed.
- copy your customized pin setting to src/pins
- Flash 2.0.x-bugfix
- Reset EEPROM 
- Set the saved values... for my it was only the M92 values and the Z offset

My TMC2208s are working fine :)

thinkyhead, 2018-09-01:
Any better with the current release 1.1.9?

boelle, 2019-02-20:
@NSchnitzler is the problem still there in latest bugfix 2.0?

NSchnitzler, 2019-02-20:
Yes Problem solved,
Thanks to all

boelle, 2019-02-20:
@NSchnitzler maybe give the green close bottom some love?

closed 2019-02-20 by thinkyhead