MarlinFirmware/Marlin #11344: heating failed. printer halted

mshavandi, 2018-07-23:
planned to make my own printer and wanted to test motors movement without connecting anything except motors and drivers and power supply. I'm using configuration: EFB with 250000 baudrate.
faced 2 problems:
1. when I want to manually control the motors, they move but just in one direction (+X). searched the google and find that they first should be homed and then they can move in both direction. So I homed all X,Y,Z axis, they moved for some time and then stopped. after that nothing changed! they can't move in both directions yet and found no answers useful in the field.
2. I loaded a model's gcode and wanted to print, hoping it could behave differently and also test the movements of the motors in action. but it keeps saying: heating failed. printer halted. I searched the google and found 3 solutions:

- writing 998 or 999 in front of `TEMP_SENSOR_0` and `TEMP_SENSOR_BED` so it always reads a constant value. did it and put 998 in front of both of them, this time the motors couldn't move even manually!(even they didn't seem to be turned on, they didn't have any noise) the same thing happened about value 999.

- writing 0 to disable sensors while testing. I did it and put 0 in front of both but it reported compile error: `TEMP_SENSOR_0` should be defined. so I put 998 in front of `TEMP_SENSOR_0` and put 0 for `TEMP_SENSOR_BED`, motors could move manually but about printing, nothing changed. kept saying: heating failed. printer halted. changed the value 998 to 999 and motors couldn't move even manually (even they didn't seem to be turned on, they didn't have any noise).

- sending M302 S0 command to enable cold extrusion. I sent M302 S0, showed: sending:M302 S0 and nothing happened. then I sent: M302 after that and showed: cold extrusion enabled. so I hit the print and then again it said: heating failed. printer halted. I tested this solution with values of 1 and 998 and 999 in front of `TEMP_SENSOR_0` but the same results happened.

I really don't know what's the problem. Can anyone help please?

Deneteus, 2018-07-23:
This same question was addressed here: 
https://reprap.org/forum/read.php?146,237827

1. You are forgetting that the firmware is looking for input from the End stops that you have currently disconnected.

End Stops
http://marlinfw.org/docs/configuration/configuration.html#endstops

2 Your Dummy Temps would have to match to keep it from triggering Thermal Protection which hasn't been disabled.

Thermal Settings
http://marlinfw.org/docs/configuration/configuration.html#thermal-settings

Dummy Thermistor Values
http://marlinfw.org/docs/configuration/configuration.html#temperature-sensors

Thermal Safety 
http://marlinfw.org/docs/configuration/configuration.html#safety

[image]

Deneteus, 2018-07-26:
For the dummy temps you have to actually configure the temp you want Marlin to show. 

In #6183 it says that they disabled the Max and Minimums not enabled. It would be easier to test everything rather than just testing the motors as there is no useful information gained by testing them other than that the motors work. 

People here are busy working on other projects and fixing bugs. I would suggest posting your question in a group meant for Building 3D printers where there are thousands of people that can answer questions.

3D Printer Builders/Designers
https://www.facebook.com/groups/cncbuilddesign/

There are also 2 Marlin Firmware groups: 

MarlinFirmware: The Official Group
https://www.facebook.com/groups/1049718498464482/

Marlin Firmware for 3D Printers User Group
https://www.facebook.com/groups/3Dtechtalk/

mshavandi, 2018-07-27:
Thanks @Deneteus, you're right. I will post my question somewhere else.

closed 2018-07-27 by mshavandi