MarlinFirmware/Marlin #11345: [Bugfix-2.0.x] [Trinamic] Axes fail to move after stopping a print.

mbuc, 2018-07-23:
### Description

I'm using TMC2130's on all four axes.  Latest library release of @teemuatlut 's TMC library.

At some point during a print cycle, one or more axes seems to stop working properly.  More accurately, it seems like the torque drops immensely to the point that almost nothing happens.

Here is an example pair of prints:

[image]

I printed this one first.  Ignore the extrusion, separation, and raft issues.  But this is what <span>#</span>2 should have looked like - the only difference was a temperature change.
   
   

[image]

Here we can see that the print completed the first 1.75 outlines of the 3-perimeter offset skirt before the first axis lost full power.  You can see that when it was drawing the final skirt perimeter it was unable to move vertically (Y axis), leaving the extra line to the left of the main body.

[image]

The Y axis failed to move after that first short portion, leaving a nice weird silhouette of the object...

[image]

Finally, after about 95 out of 229 layers the E0 axis experienced the failure.  The reasons I suggest that it is a motor current / torque issue rather than the motor being shut off are:

- The extruder continues to move to retract filament.
- The extruder unretracts filament up until it experiences resistance (the filament hitting and filling the heat chamber).
- At this point the extruder makes no noise to indicate skipping steps, and does not seem to be grinding filament, but it is unable to push any PETG through the nozzle.

### Steps to Reproduce

This issue seems to happen after a certain amount of system on time.  However, I have found a process that seems to make it happen more quickly:

1. Start a print over USB.
2. When the print starts to home the Z axis, stop the print.
3. Start a new print and wait.

Variables in my instance: Using Simplify3D 4.0 over USB to a RAMPS 1.4 board on a 2560.

**Expected behavior:** The object prints normally and all axes are able to move within their defined plane.

**Actual behavior:** At some point one or more axes is unable to move, or is unable to move more than a very short distance.

#### Additional Information

* My `Configuration.h` and `Configuration_adv.h` files: [Configuration_072318_bf2.x.zip](https://github.com/MarlinFirmware/Marlin/files/2218438/Configuration_072318_bf2.x.zip).
* See my comment [here](https://github.com/MarlinFirmware/Marlin/issues/11272#issuecomment-405659965).

DerThommi, 2018-07-26:
Hey, I'm experiencing nearly the same issues currently. I have 3 TMC2130 for my X Y Z axis as well.
When I start my printer (RADDS+Due) I can move all axis via Pronterface without any issues. When I start a print 2 or all 3 stepper drivers stop working. I tracked it down to a correlation with the hot end cooling fan. As soon as this fan starts my stepper lose connection. I can reproduce this by starting the printer, moving the axis manually via pronterface, everything works fine. If I send M106 P1 to start the fan the stepper driver stop working. Maybe it's something similar for you. 

I haven't found a solution and might open an issue as well.

mbuc, 2018-07-28:
Very interesting.  I'll try it and see.

I just got an overtemperature warning while cold on Y and E0.  I power cycled and turned on my bed, which has been the only thing on for a few hours.  Thermal protection shutdown occurred for the bed, which is odd because it was working and was at temperature 5 minutes ago.  When I hit the reset button, I got the overtemperature warnings on the drivers that are completely at room temperature (71*F, 48.8% RH).

closed 2018-11-05 by thinkyhead