MarlinFirmware/Marlin #11349: G10/G11 Fwretract when using Mixing Extruder

sergiofagundes, 2018-07-23:
Hi. When I use G10 / G11 using a mixing extruder (Diamond Hotend) the retraction uses the mixing ratio as retraction for each stepper motor instead of the full amount of retraction.

For example: If I have a 10mm retraction setup and my current tool uses a mixing ratio of 0.1/0.3/0.6 then each motor will retract as 1mm/3mm/6mm. Is there any way to make all of the motors retract the same amount of mm ignoring the tool mixing ratio ?

ejtagle, 2018-07-24:
It should work like that: Retraction using the mixing ratio. What is the purpose of equal retraction ?

sergiofagundes, 2018-07-24:
Example:
T0 ; Using a mixing ratio (1.0, 0.0, 0.0)
more gcodes here...
G10 ; end of layer retraction (on Simplify3D)
T1 ; New tool Using a mixing ratio (0.1, 0.3, 0.6)
G11 ; OPS!!! Problem, will unretract all 3 filaments, worst yeat - the 1st filament was retracted at 100% ratio but unretracted at only 10% ratio.

In repetier the G10/G11 will make an equal retraction. The purpose is to avoid ooze filament, jamming the hotend.

Mixing factors should be used for printing purposes. It would be nice to have an option to have the G10/G11 to use equal retraction or mixing ratio retraction.

tcm0116, 2018-07-30:
I started working addressing this issue a few month ago before I moved. I should hopefully be able to get back to looking at it soon.

sergiofagundes, 2018-08-01:
Hi tmc0116. Let me know if you get this issue fixed. Thank you.

boelle, 2019-02-20:
@sergiofagundes is the problem gone with latest bugfix 2.0?

tcm0116, 2019-02-20:
@boelle - this has been fixed

boelle, 2019-02-20:
then either @sergiofagundes  or @thinkyhead should give the green close botton some love :-D

sergiofagundes, 2019-02-21:
Hi. I have stopped using the diamond hotend so I can't test it. I'm using prusa mk3 mmu2 and have given up with diamond hotend.

closed 2019-02-21 by sergiofagundes