MarlinFirmware/Marlin #11352: Z position wrong 5mm above heat bead

Blisk, 2018-07-24:
### Description

Finally manage to make auto bed level working but, pritner start to print 5mm above heat bed.
When I test with G28 goes down to the bed and G29 also, but when test with G0 Z0 it goes 5mm above heatbed. I checked all solutions on internet but nothing works.

### Steps to Reproduce

1. G28 works OK
2. G29 works OK
3. G0 Z0 stop at 5mm above heatbed
4. printer prints start at 5mm above heatbed

#### Additional Information

Marlin 1.1.8

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2223637/configuration.zip)

GMagician, 2018-07-24:
I think that you need to test latest bugfix branch before someone can help you. 1.1.8 is too "old" and many bugs have been fixed in between

Blisk, 2018-07-25:
Which one is than the latest?

gloomyandy, 2018-07-25:
The latest version is either bugfix-1.1.x or bugfix-2.0.x both are currently kept in sync, only bugfix-2.0.x works on 32 bit systems, both work on AVR based systems. You should probably not just copy your configuration files over, but rather use a file comparison program and convert one of the bugfix example configurations to match your needs. There are quite a few new settings that you may need.

closed 2018-11-05 by thinkyhead