MarlinFirmware/Marlin #11353: Q: Any of marlin features that could make circle objects "fuzzy"

boelle, 2018-07-24:
i'm trying to dialing in my tolerances and i get a strange fuzzy surface on circle objects

[image]

the outside of the box is crisp and sharp

but the inside of the box is not, and the same with the dish, resulting in that they fuse together

there needs to be at least 1mm in difference of diamter and they need to be printed next to each other to be able to push them together

any clue's and pointers are welcome

i did a thin wall object to make sure i dont over or under extrude
and i have also done a pid tune

boelle, 2018-07-24:
anyways dinner time, will print the same gcode from sd card later tonight, thanks for the tip

Blisk, 2018-07-24:
exactly, I get the same, but when printing form SD card, circles are nice and smooth

boelle, 2018-07-24:
yeah... when it prints the circles do not sound like one fluid and smooth montion

boelle, 2018-07-24:
same gcode just copied to a sd card makes a huge difference

thanks for that one @Blisk  :-D

@thinkyhead are there any buffer since i could increase or am at the limit with mega+ramps?

GMagician, 2018-07-24:
Maybe you can increse RX_BUFFER_SIZE (and let octoprint knows you have a bigger buffer. Repetier has a configuration parameter)

GMagician, 2018-07-24:
With mega2560 I'm using 1024 and in repetier I set 768

Blisk, 2018-07-24:
I also have 2560 and will try this tomorrow, because I like to print it like before from PC. That worked for some time but now doesn't anymore.

boelle, 2018-07-24:
Same here, but its time to get some sleep soon

i have done firmware changes etc so printer is ready, just need to make sure i have same code on sd card and octoprint so i can compare

Roxy-3D, 2018-07-24:
If you print a sheet with a circle bored out of it...  What ever the diameter of the circle is, it will be less when it is printed.    Is that what is making stuff not fit correctly for you?

You might want to try it and use a micrometer on your printed part.

jonathanweinberg, 2018-07-25:
You might want to have a read about this issue here (I’m not the author): http://lokspace.eu/bad-print-quality-with-usb-or-octoprint-the-solution-is-here/

boelle, 2018-07-25:
@Roxy-3D nope its not the diameter that is the problem its the surface quality

when printing from sd card everything is ok

when printing from octoprint the surface gets fuzzy with a lot of micro "blobs"

boelle, 2018-07-25:
@jonathanweinberg its not spammy if its helping :-D just got up so will have breakfast first

boelle, 2018-07-25:
anyways, closing this one to reduce my footprint on the issue list, sd card printing is one fix but will have a look at  @jonathanweinberg link also

closed 2018-07-25 by boelle