MarlinFirmware/Marlin #11354: Feature Request: Pause/Continue

Dentrax, 2018-07-24:
The pause feature with the current version does not currently exist. In some cases, sometimes we have to pause printing. Then we can continue the position we paused. When we pause it, the stepper motors can rise to the highest. When we press continue, the steppers should return to the paused-positions where they saved the paused-positions to memory. Then, they continue to printing. :)

Is it possible? Or will it be such a feature?

Roxy-3D, 2018-07-24:
Have you tried the #define ADVANCED_PAUSE_FEATURE ???     It does more than what you are asking for.    You can control just about everything.     The Basic stuff can be controlled by just adjusting the :+1: ```
/**
 * Nozzle Park
 *
 * Park the nozzle at the given XYZ position on idle or G27.
 *
 * The "P" parameter controls the action applied to the Z axis:
 *
 *    P0  (Default) If Z is below park Z raise the nozzle.
 *    P1  Raise the nozzle always to Z-park height.
 *    P2  Raise the nozzle by Z-park amount, limited to Z_MAX_POS.
 */
#define NOZZLE_PARK_FEATURE

#if ENABLED(NOZZLE_PARK_FEATURE)
  // Specify a park position as { X, Y, Z }
  #define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MAX_POS - 10), 20 }
  #define NOZZLE_PARK_XY_FEEDRATE 100   // X and Y axes feedrate in mm/s (also used for delta printers Z axis)
  #define NOZZLE_PARK_Z_FEEDRATE 5      // Z axis feedrate in mm/s (not used for delta printers)
#endif
```

Dentrax, 2018-07-24:
Thank you, Exactly what file do I have to define it? Will there be a pause segment on the LCD interface after define it and flashed to Arduino?

Roxy-3D, 2018-07-24:
The Nozzle_Park_Feature is in Configuration.h.    The Advanced_Pause_Feature is Configuration_adv.h
You can control the pause, resume, and filament change from the LCD Panel.

Dentrax, 2018-07-25:
Can you tell me what was going on when if AC electricity is cut-off in an unexpected state and assume that it is not connected a power source? In this case, can it resume where it left-off? Does it have a memory controller (maybe an EPROM manager algorithm) that writes the latest stepper positions?

Dentrax, 2018-07-26:
I don't have an opinion on the subject, but do we need to do something like an EEPROM? (https://www.arduino.cc/en/Reference/EEPROM)

How reasonable is it to write dozens of data in a second to EEPROM? (I mean positions) 

Do you have an issue link about this discussion? (If it was opened before)

thinkyhead, 2018-07-26:
> Can you tell me what was going on when if AC electricity is cut-off in an unexpected state and assume that it is not connected a power source?

Marlin now has the feature `POWER_LOSS_RECOVERY` to resume SD printing after a power outage. By default it requires no extra hardware. It saves the current print progress to the SD card periodically (once per layer). But it also has support for a digital pin to support devices that can detect power-loss early, in which case it performs the save only once, when power-loss actually occurs.

FromtonRouge, 2018-08-19:
Hello @thinkyhead, what kind of device exactly do we need for the digital pin? An UPS ? I can't find documentation on it. If it's an UPS I don't understand how to connect it to the digital pin

closed 2019-02-19 by thinkyhead