MarlinFirmware/Marlin #11359: Please improve error messages for automation

AndKe, 2018-07-24:
### Description

"Error" or !! not used where it should ?

### Steps to Reproduce

if printer fail , outputting lines like this: 
[image]

it won't be detected by any smart feature like: 
[image]

**Expected behavior:** [What you expect to happen]

I would expect OctoPrint to stop

**Actual behavior:** [What actually happens]

OctoPrint continues, because there is no string like "Error:" or "!!" 
-and the original error is buried beyond buffer...

thinkyhead, 2018-07-27:
That string is standardized and should be recognized (and handled) by hosts. You might want to submit a report about this to the OctoPrint project.

AndKe, 2018-07-27:
is "Printer stopped" the string to look for , or , is there maybe a preceding message I missed, that should be handled as well ?

Thank you.  - I'll suggest that for OctoPrint once you answer.

closed 2018-07-28 by AndKe