MarlinFirmware/Marlin #11360: [Bugfix-2.0.x] Malyan M200 Compilation error: "too many arguments to function"

numanair, 2018-07-25:
### Description

I am trying to compile with PIO for my Malyan M200, which I was able to do until recently. I am still able to compile, but only with `SDSUPPORT` and `MALYAN_LCD` disabled. I was previously able to compile (with these features), presumably before c83109c.

### Steps to Reproduce

1. Compile for `BOARD_MALYAN_M200` with `SDSUPPORT` and `MALYAN_LCD`

**Expected behavior:**
Compiles successfully
**Actual behavior:** 
Returns the following error:

> Marlin\src\module\configuration_store.cpparlin\src\lcd\malyanlcd.cpp: In function 'void process_lcd_s_command(const char*)':
> 
> Marlin\src\lcd\malyanlcd.cpp:329:44: error: too many arguments to function 'bool enqueue_and_echo_command(const char*)'
> enqueue_and_echo_command("G28", false);
> ^
> In file included from Marlin\src\lcd\../module/planner.h:38:0,
> from Marlin\src\lcd\malyanlcd.cpp:49:
> Marlin\src\lcd\../module/../gcode/queue.h:98:6: note: declared here
> bool enqueue_and_echo_command(const char* cmd);
> ^~~~~~~~~~~~~~~~~~~~~~~~

#### Additional Information
Supplied as requested as I am not sure what else is needed.

xC0000005, 2018-07-25:
This is probably caused by my PR to update the LCD functionality. I'll fix this in a bit, but for now you fan remove the ", false" bits and it will likely work.

xC0000005, 2018-07-26:
This should be fixed now - sync with the latest bugfix-2.0.x and see if it works for you without other changes.

closed 2018-07-27 by thinkyhead