MarlinFirmware/Marlin #11369: Proper way to debug a TMC2130 failing?

MarioStoilov, 2018-07-26:
I've been having issues where my Extruder TMC2130 starts misbehaving. The motor start buzzing when having to do a retraction and does it half way (it pulls, but when it has to push it back it buzzes and does nothing) but continues normally for normal extrusion. This happens in prints with heavy retraction such as torture tests, large amount of pillars etc. 

At first, I thought it was because of the driver overheating, being a geared 3:1 extruder and the prewarn flag being raised. I switched from a 20x20x10 fan to a 120x120x50 fan, but that had no effect.

The funny thing is that resetting the printer fixes this, but it fails the same way if I start the print again.

So the question here is: how do I debug this? s there a way to print the TMC temperature and see what it's saying (the heatsink is cold on touch) or is this something else?

teemuatlut, 2018-07-26:
Slow down your retraction speeds.
The buzzing sounds like the driver can't keep up with the requested speed.

MarioStoilov, 2018-07-26:
I'll try that after i get back from vacation.

but at 50mm/s with 6mm length seems to me that it's already low. What does the buzzing actually mean? Too hot, too much current, something else?

DerThommi, 2018-07-26:
Take the filament out, use pronterface and start with a low speed and extrude 20mm, steadily increase the speed until you hear the buzzing sound. I can't go over 600mm/min (=10mm/s).

Might need to enable cold extrusion.

tig33r, 2018-07-26:
Extrude at least 50mm to get maximum pressure in hotend.

MarioStoilov, 2018-07-26:
I have no problem with these speeds for the test. The issue shows itself after about an hour of printing.

marcio-ao, 2018-08-14:
Check M122 (TMC_DEBUG) for any flags. We've recently had an issue where s2g was being tripped and shutting down the drivers. Make sure that isn't happening in your case.

mbuc, 2018-08-16:
@MarioStoilov I have the same issue on multiple axes about an hour to two hours into a print.  Torque drops and steps are lost.  Most recently when this happened, I reset the printer doing the usual: full shutdown, PSU on, then USB connection to the Arduino.  I homed Y and X, and a moment later I got overtemperature warnings on several chips.  I know it was at least X, Y, and E0; I'm not sure about Z.  I hadn't even sent power to E0.  The chips were completely cool to the touch, of course.  

Unfortunately, Windows 10 demonstrated its most powerful feature again last night: rebooting and updating at the worst moments possible.  If and when it happens again I'll post a log.

boelle, 2019-02-20:
@MarioStoilov still having problems?

MarioStoilov, 2019-02-20:
No. Turning down the speed solved the issue.

boelle, 2019-02-20:
then maybe click the green close button below?

closed 2019-02-20 by MarioStoilov