MarlinFirmware/Marlin #11376: [Bugfix-2.0.x][TMC2130] Driver lose connection when fan starts

DerThommi, 2018-07-26:
### Description

I'm using a RADDS+Due board with the latest bugfix (downloaded 1 hour ago) and 3 Watterot TMC2130 for XYZ axis. When I start a print the steppers stop working. After some debugging I found out that the hot end cooling fan is causing an issue. Now I'm able to reproduce the issue (although I'm not sure if the fan is the only thing causing the drivers to lose connection.

At first the M122 looks like this and I can control all axis:
`>>> m122
SENDING:M122
		X	Y	Z
Enabled		false	false	false
Set current	700	700	700
RMS current	673	673	673
MAX current	949	949	949
Run current	21/31	21/31	21/31
Hold current	10/31	10/31	10/31
CS actual		10/31	10/31	21/31
PWM scale		20	19	51
vsense		1=.18	1=.18	1=.18
stealthChop	true	true	true
msteps		16	16	16
tstep		1048575	1048575	1048575
pwm
threshold		0	0	0
[mm/s]		-	-	-
OT prewarn	false	false	false
OT prewarn has
been triggered	false	false	false
off time		5	5	5
blank time	24	24	24
hysteresis
-end		2	2	2
-start		3	3	3
Stallguard thrs	0	0	0
DRVSTATUS	X	Y	Z
stallguard
sg_result		0	0	0
fsactive
stst		X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:	X = 0x80:0A:00:00
	Y = 0x80:0A:00:00
	Z = 0x80:15:00:00

>>> m122 s1
SENDING:M122 S1
axis:pwm_scale |status_response|
X:19 |0b1000| 	Y:19 |0b1000| 	Z:29 |0b1000|
X:20 |0b1000| 	Y:20 |0b1000| 	Z:28 |0b1000|
X:19 |0b1000| 	Y:19 |0b1000| 	Z:32 |0b1000|
X:20 |0b1000| 	Y:20 |0b1000| 	Z:31 |0b1000|
X:19 |0b1000| 	Y:20 |0b1000| 	Z:29 |0b1000|
>>> m122 s0
SENDING:M122 S0
`
Then if I start the fan I lose complete control over all steppers.

`>>> m106 p1
SENDING:M106 P1
>>> m122
SENDING:M122
		X	Y	Z
Enabled		false	false	false
Set current	700	700	700
RMS current	1215	1215	1215
MAX current	1713	1713	1713
Run current	21/31	21/31	21/31
Hold current	10/31	10/31	10/31
CS actual		0/31	0/31	0/31
PWM scale		0	0	0
vsense		0=.325	0=.325	0=.325
stealthChop	false	false	false
msteps		256	256	256
tstep		1048575	1048575	1048575
pwm
threshold		0	0	0
[mm/s]		-	-	-
OT prewarn	false	false	false
OT prewarn has
been triggered	false	false	false
off time		0	0	0
blank time	16	16	16
hysteresis
-end		-3	-3	-3
-start		1	1	1
Stallguard thrs	0	0	0
DRVSTATUS	X	Y	Z
stallguard
sg_result		0	0	0
fsactive
stst		X	X	X
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:	X = 0x80:00:00:00
	Y = 0x80:00:00:00
	Z = 0x80:00:00:00
`

Using M106 P0 to start the print cooling fan doesn't cause an issue though.
Disabling TMC_DEBUG and MONITOR_DRIVER_STATUS doesn't change anything. 
I also changed the Arduino with no success.

### Steps to Reproduce

1. Start hot end cooling fan
2. Move any axis, nothing happens

DerThommi, 2018-07-27:
Any ideas? With Repetier everything works fine.

DerThommi, 2018-07-31:
Some further testing: Switching to the native USB port on the due, nothing changed besides that I have to click connect twice in pronterface. Bought a new (genuie) Due, still the issue stays, as soon as the hot end fan gets toggled on the stepper driver stop working. I also tried software SPI with pins 33, 35, 37 with no success.

landodragon141, 2018-08-02:
I think there is something funny in the code for the 2130. I have a similar problem but only with my x axis. I can't correlate mine too the fan though. That doesn't seem to effect anything. I also originally thought it was hardware but I'm positive it's the firmware now. I'm going to try and go through and disable features 1 by 1 till i hit bare bones to see if i can figure it out.

landodragon141, 2018-08-02:
I'm using a MKS Gen 1.4. The problem exists in both 1.1.9 and 2.0.x bugfix.

DerThommi, 2018-08-02:
I can finally say that I found the issue. It's the fan. The original one provided by E3D. After testing all scenarios at all available FETs I could narrow it down to the fan and I switched it out and the problem is gone.

Can you go into more detail what's your problem is? What does M122 command say?

closed 2018-09-01 by thinkyhead