MarlinFirmware/Marlin #11377: New M code for SD abort/stop?

zachwelch, 2018-07-26:
### Description

I am writing new host software to control my printers, which run Marlin bugfix-1.1.x. I have implemented SD card support for all existing M-codes, and my implementation seems to have exposed one significant shortcoming: there is no way to abort an SD card job from the host.

### Steps to Reproduce

1. Start a job from the SD card (M23, M24)
2. Watch the print fail, pause it (M25), then decide the job needs to be aborted.
3. Realize your only option is to use the printer to abort the job.

**Expected behavior:** I expect there to be a matching M-code to abort the job.

**Actual behavior:** I am forced to use the LCD menu to abort the job.

#### Additional Information

Looking at the code, the block that gets triggered when ``abort_sd_printing == true`` performs unique cleanup steps. Presently, the only place that flag gets set is in the ``ultralcd.c`` module. Thus, it seems reasonable to add an M-code whose primary purpose is to set that flag.

I can probably code up this functionality myself in an afternoon; however, I wanted to submit this issue first to confirm that this feature would be welcomed upstream (after the PR/review/revision process). 

Finally, this task also implies additional documentation (e.g. http://marlinfw.org/meta/gcode) and translations (if the new code sends a message to the host to indicate the final job disposition). I'd be happy to do what I can to address those secondary issues as well, with a little guidance from those who know how to do those things.

thinkyhead, 2018-07-27:
It's a good proposal.

zachwelch, 2018-07-27:
This appears to be a duplicate of #10295.

thinkyhead, 2018-11-05:
Duplicate of #10295

closed 2018-11-05 by thinkyhead