MarlinFirmware/Marlin #11378: [bugfix-2.0.x] bltouch stows when homing Z

hamster65, 2018-07-26:
The probe reacts fine to gcodes but when the printer homes, it does not deploy but stows. The print head crashes into the bed after that.
It worked fine yesterday and stopped working after I flashed 2018-07-26 05:58:28 -0500.

GMagician, 2018-07-26:
It shouldn't but I think is something I did with bltouch. Maybe I saw where is the problem.

GMagician, 2018-07-26:
@hamster65 try posted pr, it should fix

tig33r, 2018-07-26:
Almost destroyed my head today too cause of failed homing Z axis ;/ Version before you bltouch commit is working good.

GMagician, 2018-07-26:
@tig33r sorry, for sure it was not my intention

tig33r, 2018-07-26:
No problem :) this is preety normal when you want to test new feature in beta or even alpha phase. 👍

GMagician, 2018-07-27:
It should work now

GMagician, 2018-07-28:
May you test #11397, it should be only a cosmetic thing, if ok post there

PR #11379 (2018-07-27): [2.0.x] fix bltouch homing
Fix #11378
closed 2018-08-13 by thinkyhead