MarlinFirmware/Marlin #11381: crazy jumps while adjusting temperature

AndKe, 2018-07-26:
### Description

(Marlin 1.1.8) , please see this video: https://www.youtube.com/watch?v=eenmmp6Disw

Observe the crazy big jumps (100°C)  - similar behavior for bed temp.
The only thing I can think of could cause this, is:
ENCODER_PULSES_PER_STEP 1
ENCODER_STEPS_PER_MENU_ITEM 5

my menu navigation is kind of ok, so the encoder may have more than one pulse per step step, but how to verify ? 
in any case, the 100 degree jumps are insane :)

thinkyhead, 2018-07-27:
> how to verify ?

Start by commenting out and letting the default apply. If that doesn't work, then experiment with different values for `ENCODER_PULSES_PER_STEP`. Usually `ENCODER_STEPS_PER_MENU_ITEM` can be left set to 1.

AndKe, 2018-07-27:
@thinkyhead 
those declares must have been something from my previous printers, - commented out, and it works fine :)
Thank you.

closed 2018-07-27 by AndKe