MarlinFirmware/Marlin #11382: bltoutch and marlin for tevo tarantula

ichigo1987, 2018-07-26:
Hello,
I wanted to add to my printer (tevo tarantula) bltouch on  marlin 1.1.x easy config unfortunately, after setting everything in my opinion well, the error pops up. I've tried different values ​​each time the same error. Please help me because I've already looked at all the videos and on tt group on facebok about marlin configuration under bltouch, and I have no idea how to fix it.
My config

error
```
SanityCheck.h:898: error: static assertion failed: BACK_PROBE_BED_POSITION is outside the probe region.
static_assert(BACK_PROBE_BED_POSITION <= MAX_PROBE_Y, "BACK_PROBE_BED_POSITION is outside the probe region.");
```
offset
```
#define X_PROBE_OFFSET_FROM_EXTRUDER  0  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER  -35,5  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER  -0,77   // Z offset: -below +above  [the nozzle]
```
boundaries
```
 // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 10
  #define RIGHT_PROBE_BED_POSITION 190
  #define FRONT_PROBE_BED_POSITION 36
  #define BACK_PROBE_BED_POSITION 164
```
bed
```
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0 - XTRA_BED_LEFT
#define Y_MIN_POS 0 - XTRA_BED_BACK
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 200
#define Z_MAX_POS 200

```
sensor
```
/**
 * Z-Probe type (must be none or one of them)
 * If a Z-Probe type is selected, a Bed Leveling type other than MANUAL must be selected.
 */
#define BLTOUCH         // ANTClabs BLTouch sensor (might also work with clones)
//#define SN04          // Green sensor
//#define INDUCTIVE_NO  // Normally open inductive sensor
//#define INDUCTIVE_NC  // Normally closed inductive sensor
//#define SERVO_PROBE   // Endstop switch on rotating arm. Set servo angles!
//#define NOZZLE_PROBE	// Nozzle wired up to contact metal on bed

```
leveling
`#define BILINEAR
`

my all files

[MarlinTarantula-EasyConfig-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/2233460/MarlinTarantula-EasyConfig-1.1.x.zip)

zarthcode, 2018-07-26:
Not sure how your compiler is setup, but maybe it doesn't allow using a comma in place of a decimal? (e.g. -35,5 becomes -35.5)

Thelvaen, 2018-07-26:
X_PROBE_OFFSET_FROM_EXTRUDER & Y_PROBE_OFFSET_FROM_EXTRUDER must be integer value so please use 35 or 36.

closed 2018-07-27 by thinkyhead

ichigo1987, 2018-07-27:
When i change value y probe to 36 it complile without any problem. Thanks for your help, because i have to print a gift for my son's birthday. I am very grateful for your help.

Thelvaen, 2018-07-27:
My pleasure :)
Enjoy your printer :)