MarlinFirmware/Marlin #11385: BLTouch probe not deploying on G28

ravini-mephisto, 2018-07-27:
### Description

Using the today's date bugfix-1.1.x 
The BLTouch probe rod is not deploying during home, causing bed to crash into nozzle.

### Steps to Reproduce

**Expected behavior:**  As in all previous bugfix flashes, the BLtouch probe should deploy the rod.

**Actual behavior:** Rod stays stowed.

#### Additional Information

The output of the Debug log:

Recv: >>> G28
Recv: Machine Type: Core
Recv: Probe: BLTOUCH
Recv: Probe Offset X:30 Y:20 Z:-0.60 (Right-Back & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv: set_bltouch_deployed(0)
Recv:   current_position=(0.00, -65.00, 40.00) : setup_for_endstop_or_probe_move
Recv: > endstops.enable(true)
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -375.00, [50.00])
Recv:   current_position=(0.00, -65.00, 40.00) : sync_plan_position
Recv: <<< do_homing_move(X)
Recv: Move Away:
Recv: >>> do_homing_move(X, 5.00, [50.00])
Recv:   current_position=(0.00, -65.00, 40.00) : sync_plan_position
Recv: <<< do_homing_move(X)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(X, -10.00, 25.00)
Recv:   current_position=(0.00, -65.00, 40.00) : sync_plan_position
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 250.00
Recv: > home_offset[X] = 0.00
Recv:   current_position=(0.00, -65.00, 40.00) :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position=(0.00, -65.00, 40.00) : sync_plan_position
Recv:   current_position=(0.00, -65.00, 40.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, -412.50, [50.00])
Recv:   current_position=(0.00, 0.00, 40.00) : sync_plan_position
Recv: <<< do_homing_move(Y)
Recv: Move Away:
Recv: >>> do_homing_move(Y, 5.00, [50.00])
Recv:   current_position=(0.00, 0.00, 40.00) : sync_plan_position
[...]
Recv: <<< do_homing_move(Y)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Y, -10.00, 25.00)
Recv:   current_position=(0.00, 0.00, 40.00) : sync_plan_position
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = -65.00
Recv:  soft_endstop_max = 210.00
Recv: > home_offset[Y] = 0.00
Recv:   current_position=(0.00, -65.00, 40.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv:   current_position=(0.00, -65.00, 40.00) : sync_plan_position
Recv:   current_position=(0.00, -65.00, 40.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: Z_SAFE_HOMING >>>
Recv:   current_position=(0.00, -65.00, 40.00) : sync_plan_position
Recv:   destination=(95.00, 85.00, 40.00) : Z_SAFE_HOMING
Recv: >>> do_blocking_move_to(95.00, 85.00, 40.00)
Recv: echo:busy: processing
Recv: <<< do_blocking_move_to
Recv: >>> homeaxis(Z)
Recv:   current_position=(95.00, 85.00, 40.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(10.00)
Recv: >>> do_blocking_move_to(95.00, 85.00, 40.00)
Recv: <<< do_blocking_move_to
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -375.00, [4.00])
Recv:   current_position=(95.00, 85.00, 0.00) : sync_plan_position
[...]
Recv: echo:busy: processing

[Marlin-origin-bugfix-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/2234216/Marlin-origin-bugfix-1.1.x.zip)

GMagician, 2018-07-27:
#11378 is it's counterpart. Please test PR posted to solve it

ravini-mephisto, 2018-07-27:
That PR is for the 2.0 codebase, the referenced file (motion.cpp) does not exist in the 1.1.x file structure.

ravini-mephisto, 2018-07-27:
That resolved the issue, thanks!

ravini-mephisto, 2018-07-27:
Unfortunately now the Z axis probe direction is suddenly reversed.  I'll try to work that out after I get home again, and post a different issue if needed.

GMagician, 2018-07-27:
@ravini-mephisto also Z should now be fixed, please test

NickNonsense, 2018-07-27:
how did you resolve this, I'm having the same issue.

NickNonsense, 2018-07-27:
Wait how did you resolve this, I literally just downloaded the bugfix and having the same issue.

ravini-mephisto, 2018-07-27:
@NickNonsense 
Copy the raw contents from https://github.com/GMagician/Marlin/blob/4688ba7ecad51a904024814a29be6990e6494b63/Marlin/Marlin_main.cpp and overwrite your current Marlin_main.cpp file with it.

ravini-mephisto, 2018-07-27:
@GMagician 
Confirmed, homing and Z probe is working as expected now with the latest changes found on https://github.com/GMagician/Marlin/commit/0109dc6dd2c4b95f9940e60acf7d1aec5927d98f
Thank you!!

NickNonsense, 2018-07-27:
Yes, confirmed working again, thank you both. Although i do now have a new problem, just not sure whether its related. The second, slower probe where the probe gets released, hits the surface and stows but then immediately drops again and stays in its lower position (blue light) while it raises, seems to rise and park itself higher than before after it completes the auto home. Also does the same while selecting the level bed (Bilinear) option on the LCD screen except it finishes with probe failed during its first probe point. 

are you having any other issues @ravini-mephisto 
not sure if its a result of my config.

ravini-mephisto, 2018-07-28:
@NickNonsense No further issues on mine.  Check that your BLTOUCH_DELAY is set to 100 and adjust the Z_CLEARANCE settings to your liking (Z_CLEARANCE_DEPLOY_PROBE is set to 10 on mine, but you can lower it if you like).

NickNonsense, 2018-07-28:
Ah, yes, those sound like settings I need to tweak. Thank you @ravini-mephisto for taking the time to reply, you have possibly saved me hours of googling and head scratching.

GMagician, 2018-07-28:
May you test #11398, it should be only a cosmetic thing, if ok post there

closed 2018-07-31 by ravini-mephisto