MarlinFirmware/Marlin #11387: [bugfix-2.0.x] AXIS_IS_TMC

JackDaFlow, 2018-07-27:
Hey, 
in the newest bugfix the AXIS_IS_TMC function was implemented. Unfortunately the AXIS_IS_TMC doesn't return TRUE for all axis apart from XYZ. In the drivers.h the if conditions doesn't seem to work. E. g. #if ENABLED(Y_DUAL_STEPPER_DRIVERS) doesn't trigger although Y_DUAL_STEPPER_DRIVERS is enabled. 
I'm using TMC2130 on X, Y, Y2, Z, Z2 and E0.

teemuatlut, 2018-07-27:
The issue is because `drivers.h` is included before `Configuration_adv.h` in `MarlinConfigPre.h`.
The order can't be changed as there are certain sections that rely on the new macros.

teemuatlut, 2018-08-03:
https://github.com/teemuatlut/Marlin/tree/bf2_fix_smart_drivers

thinkyhead, 2018-08-04:
I've applied a patch, so please give the latest code a test and see if it's compiling properly.

boelle, 2019-02-20:
@JackDaFlow did you give latest bugfix 2.0 a go?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead