MarlinFirmware/Marlin #11391: PID autotune for extruder fails

maragkose, 2018-07-27:
### Description
Hi,
Installed Marlin 2.0 bugfix
Autotuning with command :  M303 E0 S200 C8 fails  when it reaches the max temp with a Thermal Runaway error. See log of printer output:

SENT:  M303 E0 S200 C8
READ: PID Autotune start
READ:  T:69.96 /0.00 B:32.97 /0.00 @:127 B@:0
SENT: M105
READ:  T:69.78 /0.00 B:32.97 /0.00 @:127 B@:0
SENT: M105
READ:  T:70.72 /0.00 B:32.97 /0.00 @:127 B@:0
SENT: M105
READ:  T:73.25 /0.00 B:32.97 /0.00 @:127 B@:0
SENT: M105
READ:  T:77.55 /0.00 B:32.97 /0.00 @:127 B@:0
SENT: M105
READ:  T:82.48 /0.00 B:32.97 /0.00 @:127 B@:0
SENT: M105
READ:  T:88.54 /0.00 B:32.97 /0.00 @:127 B@:0
SENT: M105
READ:  T:94.50 /0.00 B:32.97 /0.00 @:127 B@:0
SENT: M105
READ:  T:101.15 /0.00 B:32.58 /0.00 @:127 B@:0
SENT: M105
READ:  T:107.36 /0.00 B:32.85 /0.00 @:127 B@:0
READ:  T:114.13 /0.00 B:32.87 /0.00 @:127 B@:0
READ:  T:120.35 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:127.06 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:133.23 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:139.66 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:145.82 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:153.04 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:161.14 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:168.95 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:175.42 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:181.60 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:186.69 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:191.58 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:195.57 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:199.39 /0.00 B:32.58 /0.00 @:127 B@:0
READ:  T:0.00 /0.00 B:32.58 /0.00 @:127 B@:0
READ: Error:Thermal Runaway, system stopped! Heater_ID: 0
READ: Error:Printer halted. kill() called!
WARNING: Device unplugged while connected to port
Disconnected.

### Steps to Reproduce

1. Run M303 E0 S200 C8

**Expected behavior:** [What you expect to happen]
Autotune to complete. 
**Actual behavior:** [What actually happens]

Thermal Runaway error as soon as the max temp is reached.

#### Additional Information
 
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to v

[connfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/2236965/connfiguration.zip)

ideos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

J-charles-C, 2018-07-28:
I,

You have a NTC thermistor ? You have choose 75 for thermistor value.

best

maragkose, 2018-07-29:
Hi, 
Yes it is NTC. I tried 1 as well did not make any difference.

RoyBerntsenDesign, 2018-08-02:
I have big problems PID tuning - I get thermal runaway not able to print

Tried Thermistor 1 and 11 on a 3950 NTC 100K

maragkose, 2018-08-02:
I was getting same problems and for now I have diabled the termal runaway protection. PID tuning though managed to pass by issuing pid autotune command with 180degress instead of 200:

M303 E0 S180 C8

maragkose, 2018-09-08:
Hi, ,
Another symptom I have noticed while testing my printer is that while I am able to print fine with up to temperature 200C when I set the extruder temp (from the slicer) above 200 i.e 210C, then during the heating phase marlin reports 200 then tries to raise the temp and the display shows suddenly 0. Then it retries but same thing. It zeroes as soon as 200C are reached and cannot pass this temp. I wonder if this is related with autopid bug or is some new bug. 
Have you found anything related to temp handling?

thinkyhead, 2018-09-09:
Try increasing your `WATCH_TEMP_PERIOD` to allow more time for heat-up before an error is thrown.

maragkose, 2018-09-19:
Thanks for the hint. I actually changed the thermistor from 75 to 1 and everything seems to work ok now...

boelle, 2019-02-20:
@maragkose so if its fixed maybe its time to click the green close button below?

closed 2019-02-21 by maragkose