MarlinFirmware/Marlin #11394: 2.0.x bugfix LPC1768 TMC2208 build error on MKS SBase

hermitcrabslab, 2018-07-27:
### Description

Current 2.0x source will fail to build for LPC1768 if TMC2208 is defined/enabled on MKS SBase.
**What I did to "fix" is was to move SoftSerial and HardwareSerial libs to the include folder and fix some includes...**
For reference I will include a diff, especial for new users who don not know what to do to get it working.
**Only one TMC2028 stepper is defined for X axes** this is an example, use other free pins for other axes/extruders.
[LPC1768_Marlin_TMC2208.zip](https://github.com/MarlinFirmware/Marlin/files/2237216/LPC1768_Marlin_TMC2208.zip)

forkoz, 2018-07-28:
heh, the diff shows your specific printer configuration and then deleting 

 Marlin/src/HAL/HAL_LPC1768/HardwareSerial.cpp | 335 --------------------------
 Marlin/src/HAL/HAL_LPC1768/HardwareSerial.h   |  91 -------
 Marlin/src/HAL/HAL_LPC1768/SoftwareSerial.cpp | 329 -------------------------
 Marlin/src/HAL/HAL_LPC1768/SoftwareSerial.h   | 120 ---------

does it still build fine if you move those files and try to compile for non-TMC driver?

hermitcrabslab, 2018-07-28:
@forkoz Yes it does compile, I even removed the TMC2208Stepper lib from ini to make sure is not included at all.
Those file where deleted from current location and moved to new one, Marlin\Marlin\src\HAL\HAL_LPC1768\include\ so they get included in the build(the scope of this issue)
The diff was to show dev's the whole modifications I did(maybe I did something I was not supposed to do..) and especial to help newbie with MKS Sbase or even Smootie boards(there is one TMC2208 pull request in Smoothieware but is not as advanced as on Marlin), wish I could find one to save me the headache and time loss.

forkoz, 2018-07-28:
I guess I'll have to try it for myself. I haven't graduated to external mosfets on this board yet (breakout board cost/availability) but TMCs should work for whoever wants them.

BTW, softwareserial and hardwareserial already are in the include folder as of june 15th.

hermitcrabslab, 2018-07-28:
@forkoz for some reason git is fooling me(never liked git while I can understand how useful is for some tasks) I can see the commit [on marlin web page ](https://github.com/MarlinFirmware/Marlin/commit/3b5dec4b13033100e411421283b0554fd9f10d51) but on my cloned repo I don't have it(dumbface) 
[image] WTH?

forkoz, 2018-07-28:
no clue. It worked when I compiled once I set the pins up:
https://github.com/MarlinFirmware/Marlin/pull/11402

hermitcrabslab, 2018-07-28:
@forkoz that commit(and others) only show up on **Marlin_RTOS** branch and your **mks_trinamic**, for some reason is not on **bugfix-2.0.x**
Pissed that I didn't found your branch sooner, could save me some big amount of time, usually I check the forks, this time I didn't...
I will keep an eye on your branch, thanks for posting your findings.

p3p, 2018-07-28:
@hermitcrabslab The changes are in bugfix-2.0.x as it is the main dev branch, (Marlin_RTOS is not LPC176x compatible and is just an experiment for how we move towards RTOS compatibility in the future), @forkoz's PR will be merged into bugfix2.0.x after review.

git does have a bit of a learning curve to keep local repositories up to date (most Marlin users just download the zips)

To keep a branch up to date:
```
git fetch --all
git checkout bugfix-2.0.x
git pull origin bugfix-2.0.x
```

if you want to keep local changes intact use `git stash` to save them, and `git stash pop` to restore, although you may have to resolve merge conflicts in doing this.

hermitcrabslab, 2018-07-28:
@p3p yeah is the main reason I "love" so much git, did forgot about last one(git pull origin bugfix-2.0.x) thx for pointing out my mistake...

closed 2018-07-28 by hermitcrabslab