MarlinFirmware/Marlin #11399: [FR] M600 option to tool change

mtmello2003, 2018-07-28:
Hello,

Could be an usefull option in M600: add the option in the display to "tool change" that run a user definied gcode script.

Nowaday we have the LCD options to "continue" and "purge more".

Tks

plloppii, 2018-07-31:
Not quite sure what you are asking for, but do you mean like switching to the second nozzle to continue printing after filament runout?

thinkyhead, 2018-08-13:
I like the idea. Marlin 2.0 is getting a reorganized LCD menu layout. Adding "tool change" as a filament change option on the LCD, especially in response to a filament runout, makes sense, as you might have filament loaded in the other extruder. In fact, there is a submission in the queue which would allow you to change to the next filament spool on runout, supporting multiple inputs to a single nozzle….

plloppii, 2018-08-13:
@mtmello2003 @thinkyhead 
So I actually implemented this feature on our Gigabots already based on Marlin 1.1.9. I was able to create a separate M601 gcode called Continuous Printing After Filament Runout.
I will clean up the code and do a pull request.

thinkyhead, 2018-11-05:
@plloppii — Looking forward to it. Previous attempts have been incomplete and poor.

boelle, 2019-02-20:
@plloppii did you make an PR?

plloppii, 2019-02-20:
Not yet, I’m not done with it yet. I also haven’t had too much time to look at it.

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
There have been a couple runs at a flying tool change on runout. Id suggest a 2.1 label.

Roxy-3D, 2019-06-22:
> There have been a couple runs at a flying tool change on runout. Id suggest a 2.1 label.

Done...

thinkyhead, 2019-06-28:
We have a 2.1 Milestone. Labels are rather more general.

qwewer0, 2021-01-01:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/10951? One or the other can be closed.

InsanityAutomation, 2021-01-01:
This feature, though hard to use and needing more work, is implemented. See TOOLCHANGE_MIGRATION_FEATURE.

closed 2021-01-01 by InsanityAutomation