MarlinFirmware/Marlin #11400: 2.0.x Re-Arm no LCD output

vilsed, 2018-07-28:
Hello, dear community!

I'm experiencing an odd issue with my delta printer. Last week I've converted it to 24V power, and after first powerup the RRDGLCD just stopped working. I highly doubt that this issue might be over-voltage to the LCD, since I'm powering both 3.3V and 5V rails from separate buck converters, output voltage of which were set before connecting them to the corresponding rails. The only things changed in firmware before that boot were addition of hotend, controller fan and enclosure lighting outputs. 

First I thought that despite all logic the LCD did actually die. Ordered another one. Hooked it up. Same thing. Tried hooking up a 20x4 display (which would actually be preferable to me, since I'm still experiencing hickups on GLCD updates, despite running on LPC), using an octopus cable and double checking of which data pins I was using (as it was written about here in issues), it did not work either.

The beeper and encoder are responsive on all three displays, as it is expected, as well as SD support. I would really like having this printer able to print standalone, because of dodgy USB communication I'm experiencing.

Thank you all for help in advance.

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2237949/configuration.zip)

J-charles-C, 2018-07-28:
I,

I have the same problem. You must wiring re-arm 5V to Ramps VCC pin and servo pin. You can use en Y.

Roxy-3D, 2018-07-28:
Did you have the LCD Panel working with the Re-ARM board prior to switching to 24 volts?

The LCD Panel's on Re-ARM do require a simple cable wire change.   And it depends on whether it is a 20x4 LCD or a Graphical LCD how you change the cable.

vilsed, 2018-07-28:
@J-charles-C I have modified my RAMPS (deleted the D1 diode), so the main voltage (24V/12V) is not fed to the Re-Arm and it's voltage regulator is unused. For that I feed 3.3V to the Vcc pin on the RAMPS, which is supposedly powering up the LPC. Tried feeding 5V there, MCU did not like it. Currently feeding 5V only to the LCD Vcc pin and nowhere else. I had this setup for gods know how long, back when mega2560 lived in that printer. Only then I fed 5V to the RAMPS Vcc, not 3.3V. after switching to LPC, and not having an external 3.3V regulator, I switched back to powering peripherals from the Re-Arm's voltage regulator.

@Roxy-3D Yes, previously the LCD worked properly. Well, sort of. I think voltage regulator on the Re-Arm was being overloaded, the display was quite a bit dimmer than it was supposed to be. And yes I think back then it was fed through the Re-Arm's voltage regulator, since at that moment I did in fact had the D1 installed and fed the LCD off the Re-Arm. But in the end it shouldn't matter, right? Power rails in there are confusing...

And as for the wiring changes, yes, I'm aware of that. For GLCD - breaking out the 5V pin, and for 20x4 - rerouting D5, D6, D7 lines, if I'm not mistaken. Found that info for MKS SBase board, but since it's using the same MCU, I figured...

vilsed, 2018-08-06:
I managed to get the GLCD working by reinserting D1 into RAMPS and feeding LCD and only LCD by 5V off my little DC-DC converter, basically setting it up just like it was before the 24V conversion. But another interesting problem arose. I have two blank horizontal lines on the LCD, one in roughly the middle, and another at the bottom, both are next line up the two that are gargling while accessing SD. What could it be?

vilsed, 2018-08-15:
It turns out that the lines were caused by the LCD module itself. Sheet metal frame, afixing LCD itself to it's board was not driven home fully, thus the zebra stripe between the two wasn't having proper contact. It was just a matter of bending LCD modules' board slightly away from RRDFGSC main board and bending those sheet metal ears into proper position.

Issues resolved, closing.

closed 2018-08-15 by vilsed