MarlinFirmware/Marlin #11408: [bugfix-2.0.x] Unable to compile for DUE with any of *_DRIVER_TYPE set to TMC2208

byebyedizzy, 2018-07-29:
Marlin 2.0.0 fails to compile for Arduino DUE board when any of *_DRIVER_TYPE is set to TMC2208.

Arduino IDE output:
----------------------------
Arduino: 1.8.5 (Windows 8.1), Board: "Arduino Due (Programming Port)"

Build options changed, rebuilding all
In file included from sketch\src\module/stepper_indirection.h:66:0,

                 from sketch\src\module/stepper.h:224,

                 from sketch\src\Marlin.cpp:36:

C:\Users\user\Documents\Arduino\libraries\TMC2208Stepper\src/TMC2208Stepper.h:9:28: fatal error: SoftwareSerial.h: No such file or directory

 #include <SoftwareSerial.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Arduino Due (Programming Port).
----------------------------

As far as I see there are 2 causes:
1. Comment is Configuration.h states:
---------------
 * Specify Stepper Driver types
 * The options are used to determine driver pulse timings as well as more advanced functionality.
---------------
But specifying DRIVER_TYPE as TMC2208 does not determine any pulse timings but assumes that drivers are in "managed mode" via UART connections. And this is not always the case. So comment is misguiding, at least for TMC2208.

2. Unconditional include of SoftwareSerial.h in stepper_indirection.cpp - some boards have no SoftwareSerial implemented. And one of them is DUE.

weed2all, 2018-10-23:
Hi man...have you managed to have tmc2208 uart on arduino due?i have arduino due + radds board and i realy want to have tmc2208 uart on this...

thinkyhead, 2018-10-26:
@teemuatlut — Is this simply not possible for Due, or could it be patched in the future?

teemuatlut, 2018-10-26:
HardwareSerial is supported, but there's not (official) SoftwareSerial library for the DUE platform.
Anyone is welcome to try to get [this ](https://github.com/antodom/soft_uart)working however.

thinkyhead, 2018-11-16:
> Anyone is welcome to try to get this working however.

Put it on the TODO list…

closed 2018-11-16 by thinkyhead

weed2all, 2018-11-29:
So any news on software serial for arduino due?

teemuatlut, 2018-11-29:
No

weed2all, 2018-11-29:
😣😣😣😣

CypisPiotr, 2019-04-23:
Let me ask this - all we need is to configure soft_uart library to work with TMC2208Stepper ? 
I'm totally new to this, so be polite please.

teemuatlut, 2019-04-23:
Something along those lines, yes.