MarlinFirmware/Marlin #11411: [bugfix-1.1.x] 'TMC2208Stepper' does not name a type

yagona, 2018-07-29:
### Description

Setup:
- modified Anycubic i3 Mega
- TMC2208 at X,Y,Z,Z2 are in legacy mode (STANDALONE)
- TMC2208 at E0 is connected via SoftwareSerial RX/TX to be used in hybrid mode

Since the simplified stepper driver configuration was introduced, my configuration fails to compile successfully.

### Steps to Reproduce

I could identify three representative tests to indicate the issue:

[Test 1] wanted configuration

Configuration.h:
#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  TMC2208_STANDALONE
#define Z2_DRIVER_TYPE TMC2208_STANDALONE
#define E0_DRIVER_TYPE TMC2208

=> compilation error:

stepper_indirection.h:300: error: 'TMC2208Stepper' does not name a type
       extern TMC2208Stepper stepperE0;

[Test 2] add another TMC2208 though not connected

Configuration.h:
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  TMC2208_STANDALONE
#define Z2_DRIVER_TYPE TMC2208_STANDALONE
#define E0_DRIVER_TYPE TMC2208

=> compilation successful

[Test 3] force library to be included

Configuration.h:
#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  TMC2208_STANDALONE
#define Z2_DRIVER_TYPE TMC2208_STANDALONE
#define E0_DRIVER_TYPE TMC2208

stepper_indirection.h:
//#if HAS_DRIVER(TMC2208)
  #include <TMC2208Stepper.h>
  void tmc2208_serial_begin();
  void tmc2208_init_to_defaults();
//#endif

=> compilation successful

**Expected behavior:** [compilation successful for Test 1]

**Actual behavior:** [compilation fails when E0 is the only TMC2208 and others are in STANDALONE mode]

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2238758/Configuration.zip)

teemuatlut, 2018-07-29:
This is another case of `drivers.h` included before `Configuration_adv.h`, as it should be. But it also means they can't both depend on the other.
`driver.h` should **not** depend on any settings from `Configuration_adv.h`.

teemuatlut, 2018-08-03:
https://github.com/teemuatlut/Marlin/tree/bf1_fix_smart_drivers

yagona, 2018-08-04:
The issue is fixed in the latest bugfix-1.1.x branch. Thank you very much!

closed 2018-08-04 by yagona