MarlinFirmware/Marlin #11415: DUE/Radds with latest Marlin bugfix-2.0.x - EEPROM version mismatch+Error writing to EEPROM

vallentinc, 2018-07-30:
### Description

Due/Radds v1.5/Raps128 on latest Marlin bugfix-2.0.x gives EEPROM write error and version mismatch:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2241845/configuration.zip)
### Steps to Reproduce
M503, M502, M500, M501 + Initialize firmware from LCD

18:33:54.299 : start
18:33:54.299 : echo:PowerUp
18:33:54.299 : Marlin bugfix-2.0.x
18:33:54.299 : echo: Last Updated: 2018-01-20 | Author: (none, default config)
18:33:54.315 : echo:Compiled: Jul 29 2018
18:33:54.315 : echo: Free Memory: 83556  PlannerBufferBytes: 1664
18:33:54.595 : N1 M110*34
18:33:55.251 : echo:EEPROM version mismatch (EEPROM=? Marlin=V55)
18:33:55.251 : echo:Hardcoded Default Settings Loaded
18:33:59.119 : N2 M115*36
18:33:59.229 : FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:MendelMax I3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
18:33:59.229 : Cap:SERIAL_XON_XOFF:0
18:33:59.229 : Cap:EEPROM:1
18:33:59.229 : Cap:VOLUMETRIC:1
18:33:59.244 : Cap:AUTOREPORT_TEMP:1
18:33:59.244 : Cap:PROGRESS:0
18:33:59.244 : Cap:PRINT_JOB:1
18:33:59.244 : Cap:AUTOLEVEL:1
18:33:59.244 : Cap:Z_PROBE:1
18:33:59.244 : Cap:LEVELING_DATA:1
18:33:59.244 : Cap:BUILD_PERCENT:0
18:33:59.244 : Cap:SOFTWARE_POWER:0
18:33:59.260 : Cap:TOGGLE_LIGHTS:0
18:33:59.260 : Cap:CASE_LIGHT_BRIGHTNESS:0
18:33:59.260 : Cap:EMERGENCY_PARSER:0
18:33:59.260 : Cap:AUTOREPORT_SD_STATUS:0
18:33:59.260 : Cap:THERMAL_PROTECTION:1
18:33:59.385 : N4 M114*35
18:33:59.385 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
18:33:59.400 : N5 M111 S6*98
18:33:59.400 : echo:DEBUG:INFO,ERRORS
18:33:59.400 : N6 T0*60
18:33:59.400 : echo:Active Extruder: 0
18:33:59.400 : N7 M20*22
18:33:59.416 : Begin file list
18:33:59.416 : End file list
18:33:59.416 : N8 M80*19
18:33:59.416 : echo:Unknown command: "M80"
18:33:59.431 : N9 M220 S100*104
18:33:59.431 : N10 M221 S100*81
18:33:59.431 : N11 M111 S6*87
18:33:59.447 : echo:DEBUG:INFO,ERRORS
18:33:59.447 : N12 T0*9
18:33:59.447 : echo:Active Extruder: 0
18:33:59.447 : N13 M155 S1*82
18:34:14.563 : N14 M503*16
18:34:14.563 : echo:  G21    ; Units in mm (mm)
18:34:14.579 : echo:  M149 C ; Units in Celsius
18:34:14.579 : echo:Filament settings: Disabled
18:34:14.579 : echo:  M200 D1.75
18:34:14.579 : echo:  M200 D0
18:34:14.579 : echo:Steps per unit:
18:34:14.579 : echo:  M92 X1600.00 Y1600.00 Z12800.00 E9398.00
18:34:14.579 : echo:Maximum feedrates (units/s):
18:34:14.595 : echo:  M203 X300.00 Y300.00 Z5.00 E25.00
18:34:14.595 : echo:Maximum Acceleration (units/s2):
18:34:14.595 : echo:  M201 X3000 Y3000 Z100 E10000
18:34:14.595 : echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
18:34:14.610 : echo:  M204 P3000.00 R3000.00 T3000.00
18:34:14.610 : echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
18:34:14.626 : echo:  M205 B20000 S0.00 T0.00 X5.00 Y5.00 Z0.30 E5.00
18:34:14.626 : echo:Home offset:
18:34:14.626 : echo:  M206 X0.00 Y0.00 Z0.00
18:34:14.626 : echo:Unified Bed Leveling:
18:34:14.626 : echo:  M420 S0 Z0.00
18:34:14.626 : Unified Bed Leveling System v1.01 inactive.
18:34:14.688 : Active Mesh Slot: -1
18:34:14.688 : EEPROM can hold 18 meshes.
18:34:14.688 : echo:Material heatup parameters:
18:34:14.688 : echo:  M145 S0 H180 B70 F0
18:34:14.688 : echo:  M145 S1 H240 B110 F0
18:34:14.688 : echo:PID settings:
18:34:14.704 : echo:  M301 P22.20 I1.08 D114.00
18:34:14.704 : echo:Z-Probe Offset (mm): (mm):
18:34:14.704 : echo:  M851 Z0.26
18:34:21.256 : N15 M502*16
18:34:21.256 : echo:Hardcoded Default Settings Loaded
18:34:26.981 : N16 M500*17
18:34:27.355 : echo:Error writing to EEPROM!
18:34:27.714 : echo:Error writing to EEPROM!
18:34:28.089 : echo:Error writing to EEPROM!
18:34:28.463 : echo:Error writing to EEPROM!
18:34:28.822 : echo:Error writing to EEPROM!
18:34:29.196 : echo:Error writing to EEPROM!
18:34:29.571 : echo:Error writing to EEPROM!
18:34:29.929 : echo:Error writing to EEPROM!
18:34:30.304 : echo:Error writing to EEPROM!
18:34:30.678 : echo:Error writing to EEPROM!
18:34:31.053 : echo:Error writing to EEPROM!
18:34:31.427 : echo:Error writing to EEPROM!
18:34:31.786 : Printer reset detected - initalizing
18:34:31.786 : start
18:34:31.786 : echo: Watchdog Reset
18:34:31.786 : Marlin bugfix-2.0.x
18:34:31.801 : echo: Last Updated: 2018-01-20 | Author: (none, default config)
18:34:31.801 : echo:Compiled: Jul 29 2018
18:34:31.801 : echo: Free Memory: 83556  PlannerBufferBytes: 1664
18:34:31.973 : N1 M110*34
18:34:32.753 : echo:EEPROM version mismatch (EEPROM=? Marlin=V55)
18:34:32.753 : echo:Hardcoded Default Settings Loaded
18:34:36.622 : N2 M115*36
18:34:36.747 : FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:MendelMax I3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
18:34:36.747 : Cap:SERIAL_XON_XOFF:0
18:34:36.747 : Cap:EEPROM:1
18:34:36.747 : Cap:VOLUMETRIC:1
18:34:36.747 : Cap:AUTOREPORT_TEMP:1
18:34:36.747 : Cap:PROGRESS:0
18:34:36.747 : Cap:PRINT_JOB:1
18:34:36.747 : Cap:AUTOLEVEL:1
18:34:36.747 : Cap:Z_PROBE:1
18:34:36.747 : Cap:LEVELING_DATA:1
18:34:36.747 : Cap:BUILD_PERCENT:0
18:34:36.747 : Cap:SOFTWARE_POWER:0
18:34:36.747 : Cap:TOGGLE_LIGHTS:0
18:34:36.747 : Cap:CASE_LIGHT_BRIGHTNESS:0
18:34:36.747 : Cap:EMERGENCY_PARSER:0
18:34:36.747 : Cap:AUTOREPORT_SD_STATUS:0
18:34:36.762 : Cap:THERMAL_PROTECTION:1
18:34:36.778 : N4 M114*35
18:34:36.793 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
18:34:36.793 : N5 M111 S6*98
18:34:36.809 : echo:DEBUG:INFO,ERRORS
18:34:36.809 : N6 T0*60
18:34:36.825 : echo:Active Extruder: 0
18:34:36.825 : N7 M20*22
18:34:36.825 : Begin file list
18:34:36.825 : End file list
18:34:36.825 : N8 M80*19
18:34:36.856 : echo:Unknown command: "M80"
18:34:36.856 : N9 M220 S100*104
18:34:36.856 : N10 M221 S100*81
18:34:36.856 : N11 M111 S6*87
18:34:36.871 : echo:DEBUG:INFO,ERRORS
18:34:36.871 : N12 T0*9
18:34:36.871 : echo:Active Extruder: 0
18:34:36.871 : N13 M155 S1*82

**Expected behavior: M500, M501, M502 and M503 to work from Pronterface/S3D/Repetier host
**Actual behavior: M500 and M501 do not work at all. 
M501:
18:44:13.854 : echo:EEPROM version mismatch (EEPROM=? Marlin=V55)
18:44:13.854 : echo:Hardcoded Default Settings Loaded

M500:
18:45:19.312 : N17 M500*16
18:45:19.686 : echo:Error writing to EEPROM!
18:45:20.061 : echo:Error writing to EEPROM!
18:45:20.419 : echo:Error writing to EEPROM!
18:45:20.794 : echo:Error writing to EEPROM!
18:45:21.168 : echo:Error writing to EEPROM!
18:45:21.527 : echo:Error writing to EEPROM!
18:45:21.901 : echo:Error writing to EEPROM!
18:45:22.276 : echo:Error writing to EEPROM!
18:45:22.635 : echo:Error writing to EEPROM!
18:45:23.009 : echo:Error writing to EEPROM!
18:45:23.383 : echo:Error writing to EEPROM!
18:45:23.758 : echo:Error writing to EEPROM!
18:45:23.992 : Printer reset detected - initalizing
18:45:23.992 : start
18:45:24.007 : echo: Watchdog Reset
18:45:24.007 : Marlin bugfix-2.0.x
18:45:24.007 : echo: Last Updated: 2018-01-20 | Author: (none, default config)
18:45:24.007 : echo:Compiled: Jul 29 2018
18:45:24.007 : echo: Free Memory: 83556  PlannerBufferBytes: 1664
18:45:24.148 : N1 M110*34
18:45:24.959 : echo:EEPROM version mismatch (EEPROM=? Marlin=V55)
18:45:24.959 : echo:Hardcoded Default Settings Loaded

UBL works fine until I try to save the results into EEPROM, then watchdog reset happens and...

Thank you,  
Valentin

Sanitar777, 2018-07-31:
Hi!
I had the same problem on RADDS + DUE.
I solved it by flashing the Repetier-firmware first and then back to Marlin.

Good luck.

vallentinc, 2018-07-31:
Thank you, I will try it (again!) when I get home.
But, I am pretty sure that I had Repetier on it before I migrated to Marlin - and that's because of the shitty autolevel routine of Repetier fw, ending in many, many errors. 
ABL in Marlin just works, and that's that! 
Eh, I will take your advice and if it works, I owe you a keg :)
Thanks, again!

Sanitar777, 2018-07-31:
Sorry... I forgot to say when you are going to flash Repetier, then you need to change the EEPROM mode ... for example, with mode '1' on mode '2' in the following parameter:

#define EEPROM_MODE 1

vallentinc, 2018-07-31:
Will do! Thank you!
Up until now, step 1 is done.
19:26:14.685 : Detected EEPROM version:19
19:26:14.685 : Transformation matrix: 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
19:26:32.604 : N14 M502*17
19:26:33.108 : SelectExtruder:0
19:26:33.111 : Info:Configuration reset to defaults.
19:26:37.004 : N15 M500*18
19:26:37.523 : Info:Configuration stored to EEPROM.
Just for giggles, i tried aG28 and G29 and the tipical messaje pops out:
19:28:52.740 : Error:z-probe triggered before starting probing.

Step 2:
Flashed Marlin and...
19:34:14.888 : Error:EEPROM datasize error.
19:34:17.727 : echo:Hardcoded Default Settings Loaded
19:34:21.035 : echo:Error writing to EEPROM!

But! Something changed. I no longer get EEPROM version mismatch (EEPROM=? Marlin=V55), instead, I get the following:
19:33:38.166 : Error:EEPROM datasize error.
19:33:38.166 : echo:Index: 603 Size: 663
19:33:38.166 : echo:Hardcoded Default Settings Loaded
I am out of ideas :(

Sanitar777, 2018-07-31:
Reflash the firmware Repetier changing parameter EEPROM several times. Get Repetier to work stably. Only then, change to Marlin. At me all has turned out.

vallentinc, 2018-08-01:
I know...
Now you probably get Error writing to EEPROM! too. 
And/or eeprom datasize error Index: 603 Size: 663 or "READ: Error:Field home_offset mismatch."
Until someone get a fix for this in Marlin, I will stop reflashing between fw and versions. From what I understand, there are a finite numbers of flashings for the I2C chip until it bricks, and I don't fancy changing it from the board right now.

idrisms, 2018-08-02:
I GOT THE SAME ERROR..

start
echo: External Reset
Marlin 1.1.9

echo: Last Updated: 2018-07-31 | Author: (none, ,......)
echo:Compiled: Aug  2 2018
echo: Free Memory: 3839  PlannerBufferBytes: 1232
echo:EEPROM version mismatch (EEPROM=? Marlin=V55)
echo:Hardcoded Default Settings Loaded

vallentinc, 2018-08-03:
@Roxy-3D, @thinkyhead - any ideas? 
Thank you

Sanitar777, 2018-08-04:
Hello All!

Just do not laugh ... :-)
Today I won this error as follows:
1. Downloaded MK4due to4.3.6
2. Filled the firmware into the printer and executed the commands M502 M500
3. I downloaded the current Marlin v2.0.x firmware and reflashed my printer.

I saw the same warning:
`EEPROM version mismatch (EEPROM =? Marlin = V55)`

But after he gave commands to the M502 M500 from the console and ... bingo! 
I saw the following in the log:
```
00: 46: 45.414: N4 M114 * 35
00: 46: 45.450: N5 M111 S6 * 98
00: 46: 45.970: echo: V55 stored settings retrieved (675 bytes; crc 36949)
00: 46: 45.974: echo: G21; Units in mm (mm)
00: 46: 45.974: echo: M149 C; Units in Celsius
```
I do not know how long this will work... :-)
Good luck!

JXPA, 2018-08-11:
Hi All.

Is not the problem still solved?
I am using Ramps-FDv2 but encountered the same error.
This is my solution.

1. Flash a Repetier-Firmware and overwrites EEPROM.
2. Start up Repetier-Host and connect to the printer.
3. Open "Firmware EEPROM Configuration"　[Alt + E].
4. Set the value of all items to 0.
5. Flash a Marlin and overwrites EEPROM.
6. done :)

I think that this happens because the value inappropriate for Marlin is saved in EEPROM.
So I tried to set the value of all items to 0.
0 is the initial value of unused EEPROM.

If the problem still does not solve it, try it.

amrithmmh, 2018-10-22:
i have the same problem inmarlin 1.1.9 (mks gen 1.4 board)
the workaround works but is it permanent?

boelle, 2019-02-20:
@vallentinc still having the issue when using latest bugfix 2.0?

vallentinc, 2019-02-20:
Yes. The same.
Gone back to Repetier, because it simply works without so much fuss.
Tried a brand new board, got the same issue with Marlin bugfix 2.0.
I'll wait for a more mature implementation.
It's been months since I raised this problem and no one from dev team cared :(
No one! That says a lot

amrithmmh, 2019-03-02:
I think you are supposed to give save eeprom and then save eeprom
Also reset the board after flashing

vallentinc, 2019-03-02:
I Did that. Read initial description. M502, M500, M501. Also tried from LCD...

amrithmmh, 2019-03-02:
but i did what @Sanitar777  suggested and it worked for me (not a fix but a workaround)

shitcreek, 2019-05-14:
any update? @vallentinc

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-09 by shitcreek

angeldpino, 2020-03-20:
I solved it by Restore failsafe, Saving to EEPROM and then Loading from EEPROM using the LCD menu