MarlinFirmware/Marlin #11418: help setting up my printer?

cyndernight, 2018-07-31:
hi there

log time user, first time installer-setup

how do i customize marlin firmware in a few ways, and can someone sit down with me and help me understand how to edit this to do what i wanna make it do?

thanks <3

emartinez167, 2018-07-31:
Go to www.marlinfw.org and follow the instructions, or go to the reprap forum.

cyndernight, 2018-07-31:
hi

thanks for your email...

i was hoping for someone to sit down and help me work trough it,  ive seen the marlin instructions and 900 pages of info isn't helpful when i have little understanding of what im looking at here

EG:    http://marlinfw.org/docs/configuration/configuration.html#Firmware-Info

mentions a custom boot screen, great...  doesn't tell me HOW to use it, WHERE it is, what kind of format i need.. or how to make it work...

you can give me a manual to a Boeing 747, don't mean i can use it to fly a plane...

metacollin, 2018-08-10:
> mentions a custom boot screen

No, it doesn't.  No where does it mention anything about a custom boot screen.  The section you linked is simply referring to the Marlin boot screen, and it tells you exactly how to use it and make it work.  Making a custom one is not supported and requires actually writing/changing source code. 

> i was hoping for someone to sit down and help me work trough it, ive seen the marlin instructions and 900 pages of info isn't helpful when i have little understanding of what im looking at here

If you can't put in the effort to learn something as simple as editing a couple of text files using extremely clear instructions, why do you think some stranger is going to spend the effort to do it for you?  

> you can give me a manual to a Boeing 747, don't mean i can use it to fly a plane...

And no one is expecting you to.  To continue your analogy, you've inexplicably come here to fly a plane even though you don't actually know how, and when handed the Boeing 747 flight manual, of course it does you no good.  This your fault though, because who would expect to learn how to fly a plane just from an operation manual? It's expected that you already know how to fly a plane.  Go learn how first, then come back.  

But that's not a great analogy because it does not represent the difficulty or complexity here.  You're not being asked to fly a plane.  You're being asked to _**edit two text files using documentation that walks you through editing them.**_  

The instructions tell you exactly which files to edit right at the top.  It tells you what each thing does and what to change if you need to.  There is no possible way it could be more clear.  The real problem here is you want someone to do it for you. 

Understand that github issues are for reporting software bugs, they are not your personal help hotline and it is extremely disrespectful and inappropriate to start an issue because you need help.  And you don't even need help, you just want someone to do it for you.  

You may be able to find someone willing to help you on the forums if you drop the entitled no-effort attitude and actually try first and have specific questions as to what you don't understand about the documentation.   If not, well, you should probably stick to the firmware that came with your printer as it doesn't require any configuration and will be much simpler and easy for you to use (download it and flash it usually).

AnHardt, 2018-08-10:
For a custom bootsreeen see an example in https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/config/examples/BQ/Hephestos_2
For a **basic** configuration, to get your first print, see https://www.youtube.com/watch?v=3gwWVFtdg-4 and use some intelligence.
For a basic general understanding of 3D-printers and there configuration follow the links at the beginning of our Configuration.h files.
 https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L46-L56

Explaining all possible details to you here will take some hundred messages.
If you have a specific question come back and ask.

cabbagecreek, 2018-08-13:
If you have a previous version on Marlin that works and wants to migrate to a newer version the Marlin3DprinterTool and the sub program MarlinConfigurator might help you.

It tries to migrate old values from the old working FW to the newer version of Marlin using a GUI.
You also get the manual page for each feature when you hover over the field in the GUI.
 
At this stage it only helps with configuration.h but that is almost 50% of the work.

cyndernight, 2018-08-13:
thank you to @AnHardt and @cabbagecreek who responded nicely.

@cabbagecreek  sadly i dont have an old FW i want to keep settings from, but thank you for your help <3

@AnHardt   i have a more narrow idea of what i would like to try...
i want to get the main board to use servo2_pin and servo3_pin (digital pin D4 & D5) to set them high or low to toggle a relay on or off

i want to have two background "if" statements working..
1: "if hotbed or hotend > 30C set servo2_pin high...  if else, set servo2_pin low"
this will give me a "heating / printing in progress / hot don't touch" kinda warning.

2: the other one is a "panic light"
"if (panic, warning, error, over temp, under temp, lockup, whatever) then set servo3_pin HIGH... else, set servo3_pin LOW"
so i can get a "this shits gone wrong" light.

need to know how to make it do this, and where to put it into the code....

cyndernight, 2018-08-13:
edit - didn't mean to close it, finger slipped

thinkyhead, 2018-08-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

closed 2018-08-13 by thinkyhead

cyndernight, 2018-08-13:
@thinkyhead i am unable to use Facebook, and the other two forums require me to make yet another bloody account in order to talk...

thinkyhead, 2018-08-13:
Sorry for the inconvenience, but we're overwhelmed here.