MarlinFirmware/Marlin #11420: Bug BlTouch - Nozzle crash in bed glass

nacho87, 2018-07-31:
I configure the bl touch/3d touch in the new version of marlin and in the bugfix 2 but the Deploy and stow of the 3d touch work only in the menuin the lcd but if use Auto - Home option in the lcd ,the 3d touch don't deploy the sensor and the nozzle crash in the bed.

Thanks

GMagician, 2018-07-31:
Please post your configuration

nacho87, 2018-07-31:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2247382/Configuration.zip)

Is the configuration.h that appear in "example_configurations\Geeetech\Prusa i3 Pro B\bltouch" with some modifcations in the steps

GMagician, 2018-08-01:
Try enabling Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN as I think this is what you have and use zmin input for bltouch.

nacho87, 2018-08-01:
@GMagician Great, thanks works again , thank you. sorry for bother with this issues. If you want i can send on PR for change the file "example_configurations\Geeetech\Prusa i3 Pro B\bltouch" or add one comment in the read me. 

You can close this

GMagician, 2018-08-01:
@nacho87 I can't close it, you have to do it

closed 2018-08-01 by nacho87