MarlinFirmware/Marlin #11421: Power Loss Recovery Crashing Marlin

plloppii, 2018-07-31:
Hi All,

Thank you to everyone who is contributing to Marlin and making it better every day!

I am currently taking a shot at implementing Power Loss Recovery into my machine. I updated to the latest version of bugfix-1.1.8. I enabled power loss recovery, however when ever I try to select a SD card file, the machine crashes. Is anyone else experiencing this? This is output of what I am seeing in Simplify3D: 

`SENT: M105
READ: ok T:29.30 /0.00 B:36.07 /0.00 T0:29.30 /0.00 T1:25.94 /0.00 @:0 B@:0 @0:0 @1:0
READ: echo:enqueueing "M23 to5bc0~1.gco"
enqueueing "M23 to5bc0~1.gco"
READ: echo:enqueueing "M24"
enqueueing "M24"
READ: echo:Now fresh file: 
Now fresh file: 
READ: open failed, File: .
SENT: M105
SENT: M105
READ: start
READ: echo:Marlin 1.1.8
Marlin 1.1.8
READ:
`

I have been doing some poking around win the M24 code because I see that it is failing there, however I have not had any success. 

Any input would be greatly appreciated!

Regards,
Noah

wailord284, 2018-08-01:
Hi I am also having the same issue. But for me it seems with or without Power Recovery causes crashing in the latest 2.0.x with large prints. Not sure why..

plloppii, 2018-08-01:
@wailord284 Yeah, I find it really hard to debug issues when the program crashes. Does anyone have any input on how we can effectively find out whats going on if Marlin crashes? I am presuming it has something to do with memory.

wailord284, 2018-08-01:
It's interesting because I'm able to print smaller objects on my delta (total build of 375) but when I print anything bigger than say 200, itll just crash right after printing the skirt.

Grogyan, 2018-08-01:
That could indeed be true, if the recovery mode is set to be triggered by Z moves

thinkyhead, 2018-11-05:
Duplicate of #11072

closed 2018-11-05 by thinkyhead