MarlinFirmware/Marlin #11422: G26 not compatible with volumetric extrusion

tcm0116, 2018-08-01:
### Description

Executing G26 with volumetric extrusion enables results in under extrusion.

### Steps to Reproduce

1. Enable volumetric extrusion
2. Run G26

**Expected behavior:** Validation pattern printed successfully

**Actual behavior:** Validation pattern printed under extruded

Roxy-3D, 2018-08-01:
I wonder what the problem can be?   The line drawing in G26 is very simple.   There are some constants that control how much filament is put out based on nozzle size and filament diameter.   But it doesn't seem like volumetric extrusion should affect it very much.

tcm0116, 2018-08-01:
I suspect it's because when volumetric extrusion is enabled, the planner expects the e distance to have the units mm^3 as opposed to the standard mm when volumetric extrusion is disabled.

I'm fairly certain the solution is to set `g26_filament_diameter` to `2*sqrt(1 / pi)` when `planner.filament_size[active_extruder] != 0`. I think you'd also want to ignore the `F` parameter to `G26` as well.

I also noticed that there is code in `G26` for retracting the filament if moves are longer than 2mm, but I never saw that working. In my case, I have `FWRETRACT` enabled, so I would actually like to see `G26` use `FWRetract.retract`. This will allow `G26` to utilize my pre-configured retraction/z-hop settings (and resolve the following todo).

https://github.com/MarlinFirmware/Marlin/blob/a2ac7e1407136928a05d9451ba813e164ed46b83/Marlin/G26_Mesh_Validation_Tool.cpp#L368

Finally, `G26` was painfully slow. It'd be nice if the print and travel speeds could be defined in Configuration.h as opposed to using `PLANNER_XY_FEEDRATE() / 10.0` (which equates to 30 mm/s with the default max X/Y feedrates in Marlin).

ManuelMcLure, 2018-08-01:
Actually, that comment about "bump height" is for the .5mm Z-bump that occurs when we do a travel move - it isn't referencing the amount of retraction. The changes to use firmware retraction would be in `retract_filament()` and `recover_filament()` and should be nicely encapsulated.

tcm0116, 2018-08-01:
@ManuelMcLure firmware retraction has Z-hop, which is analogous to G26's Z-bump, so having both would be redundant.

ManuelMcLure, 2018-08-01:
@tcm0116 You're right - I just dug a little deeper and noticed that. That complicates things a bit since there are places in `G26` where we call `retract_filament()` without doing an explicit hop (such as in `prime_nozzle()`). However, I think that may not be a problem as long as we keep track of whether we're "hopped" or not in the non-`FWRETRACT` case. We could add the explicit hop code to `retract_filament()` and `recover_filament()` so in the non-`FWRETRACT` case we get similar behavior to the `FWRETRACT` case.

tcm0116, 2018-08-01:
A simpler approach might be just to always lift Z when retracting. You could move that into `retract_filament`, which makes using firmware retraction trivial.

Roxy-3D, 2018-08-01:
> A simpler approach might be just to always lift Z when retracting. You could move that into retract_filament, which makes using firmware retraction trivial.

I like this idea...     But the other idea is good too  (keeping track of whether we're "hopped" or not).

tcm0116, 2018-08-01:
If we hop at the same time as we retract, then you just have to keep track
of one state. Then, if firmware retraction is enabled, the retract/hop can
be completely passed off to the FWRetract class.

Roxy-3D, 2018-08-02:
Good thinking...   And that will save a little bit of time compared to having two separate actions.