MarlinFirmware/Marlin #11423: filament sensor on marlin 1.19

stefi01, 2018-08-01:
I transfered all the settings from marlin 1.18 to 1.19 and i get an error with the filament width sensor.
works as it should without problem in 1.18 and the settings were double checked against 1.18

temperature.cpp:1067: error: 'FILWIDTH_ERROR_MARGIN' was not declared in this scope

     if (ABS(filament_width_nominal - filament_width_meas) <= FILWIDTH_ERROR_MARGIN)

                                                              ^

sketch\temperature.cpp: In static member function 'static void Temperature::readings_ready()':

temperature.cpp:1752: error: 'raw_filwidth_value' was not declared in this scope

     current_raw_filwidth = raw_filwidth_value >> 10;  // Divide to get to 0-16384 range since we used 1/128 IIR filter approach

                            ^

GMagician, 2018-08-01:
Post configuration.h and configuration_adv.h. Missing define should be in latter file

stefi01, 2018-08-01:
Here are the files, github would not let me upload the original file format so converted them.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2251323/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2251324/Configuration_adv.txt)

Scratchbuild52, 2018-08-09:
I have the same problem. I saw it is declared but in a different place than 1.18. Can you please look at this.

Regards

thinkyhead, 2018-09-12:
Now patched in the bugfix branches.

closed 2018-09-12 by thinkyhead

ve3sjk, 2018-09-25:
Can you tell me where to find the fixed files i just turned on the Filament sensor and i am also receiving the error: 'FILWIDTH_ERROR_MARGIN' was not declared in this scope and the display is showing zero for filament reading with 1.80 volts on A5

stefi01, 2018-09-26:
i am getting the same error, coppied exactly as it was in 1.18 so i stayed
at 1.18 as it works with that version

ve3sjk, 2018-09-26:
Thanks for the reply I will download 1.1.8 from the repository
And do a comparison against 1.1.9 and try and make the
Changes needed to get it working in the newest version.

ve3sjk, 2018-09-27:
I posted a way to fix the compile error in 1.1.9 to the open ticket
On github. Basically moving 3 lines of code in the temperature.cpp
File. This allowed compile and seems to get the sensor working.

Regards,
Steve

stefi01, 2018-09-28:
I downloaded the latest bug fix marlin, transferred all the settings yet again from 1.1.8 to 1.1.9 without any further changes and it compiled and uploaded with out problem, all works now 👍