MarlinFirmware/Marlin #11427: Creality3d CR-2020 motherboard support

sandro197802, 2018-08-01:
creality cr-2020 motherboard support in bugfix 2.0.x

I don't see anything about cr-2020 in Marlin supported boards, is it possible to support it?

sandro197802, 2018-08-03:
I asked it and received the .hex and, as soon as I understand, it's useless to extract the pinout of the board.
I have also requested a workable version without having a positive answer

boelle, 2018-08-03:
well it seems that we have to start shout that they have gone closed source again

sandro197802, 2018-08-03:
I'm thinking about flashing a ramps configuration on that board, I think it's a custom version with standard pinout. I'll check some pins to have confirmation

boelle, 2018-08-03:
yeah you need a 100% trustworthy pin file at least

you will also need to put on a bootloader i think

sandro197802, 2018-08-03:
Having the pinout will be the best, a pin.h good too. I also believe it hasn't the bootloader, this is another problem😓

sexycyborg, 2018-08-03:
Have asked every day this week...doing my best.

It's quite difficult now.

boelle, 2018-08-03:
burning a bootloader is not the biggest issue, there are tons of examples on how to do that

but yeah, the board needs to defined in arduino ide so that you can tell it the correct bootloader to program on to the board

boelle, 2018-08-03:
@sexycyborg 

no one demands more than your best, but you are the only one the world knows that have contact with Creality and that knows both the western world and how business is done in china

so you are a very famous person :-D

sandro197802, 2018-08-03:
@sexycyborg

Thank you so much for the effort you're putting to make these two worlds a bit closer.
Keep us informed about any news

sandro197802, 2018-08-03:
@boelle burning it is easy but without a pinout I'n not sure about wiring between this board and Arduino uno

boelle, 2018-08-03:
both boards will have a 6 pin connector and for burning just the bootloader you dont need a pinout, its a international standard

[image]

sandro197802, 2018-08-03:
Wow, thanks a lot 👌👌

boelle, 2018-08-03:
most likely you will find a 6 pin somewhere on the creality board

and with a multimeter you can quickly find pin 2 and 6 :-D it should give you 5v when powered

boelle, 2018-08-03:
btw how does the board inside the printer look like?

sandro197802, 2018-08-03:
I found it and I can read pin numbers on the pcb 👍🏻

sandro197802, 2018-08-03:
[image]
[image]

boelle, 2018-08-03:
this is just an example guide: https://www.instructables.com/id/Flashing-a-Bootloader-to-the-CR-10/

it might be a different board but most of it will be the same

boelle, 2018-08-03:
sorry for asking so much, but what model printer is it? i might be able to find a guide for your precise printer

sandro197802, 2018-08-03:
It's a creality3d cr-2020, nothing found on the web

boelle, 2018-08-03:
this one: https://www.gearbest.com/3d-printers-3d-printer-kits/pp_608133.html

?

sandro197802, 2018-08-03:
I like this board, it looks robust and well designed, but its original firmware is too poor. I don't want to change the board with another, I rather send the printer back and buy a Wanhao D6 or a monoprice maker ultimate

sandro197802, 2018-08-03:
Yes, that one

boelle, 2018-08-03:
can you see what text is on the big chip in the middle of board?

but yes looks like a nice printer

we need to make sure we put on the correct bootloader for this board

sandro197802, 2018-08-03:
It's a mega2560

sandro197802, 2018-08-03:
But burn bootloader without a firmware to flash is useless right now

sandro197802, 2018-08-03:
What's the best way to map pins on a board?

boelle, 2018-08-03:
yes but we can prepare as much as possible

boelle, 2018-08-03:
hmm.... the slow but best way is to track wires from chip to stepper drivers and fet's(transistors)

sandro197802, 2018-08-03:
Does a multimeter help?

boelle, 2018-08-03:
nope.... you have to look at pin one on big chip, where does that go.... then pin 2....

very slow work but its the most precise way

of course we can hope @sexycyborg has more luck

boelle, 2018-08-03:
anyways... its getting late here so i will go to bed very soon

will be arround tomorrow

sandro197802, 2018-08-03:
Thanks for your help, 
Have a good night😉

silentninja1, 2018-08-04:
FYI, I have a nearly working build on one of these I had started awhile ago. IIRC all I was missing was the power enable pin. Ill dig it back up and go over it again soon.

houseofbugs, 2018-08-09:
@silentninja1 can you share the pins for that board? Been trying to get Creality to release source and it's falling on deaf ears.

houseofbugs, 2018-08-09:
Also that 2560 should NOT need the bootloader. I don't have that board to test but typically only the 1284P boards need the bootloader burned.

Roxy-3D, 2018-08-09:
> FYI, I have a nearly working build on one of these I had started awhile ago. IIRC all I was missing was the power enable pin. Ill dig it back up and go over it again soon.

If somebody has a machine with the Power Enable Pin wired up....   It should not be too hard to find out what pin it is on using M43.     Even an LED can be used to find the correct pin number.   And at that point, you could add it to your almost correct Configuration.h files and have a reverse engineered setup.

silentninja1, 2018-08-12:
Ive talked to Tim a bit offline here, but neither of us has actual hardware to find the last remaining pin with. Ill see about getting something in hand at some point to finish following it back.

sandro197802, 2018-08-13:
@sexycyborg Am I too optimistic hoping in a creality source code release?

thinkyhead, 2018-08-21:
@sandro197802 — Keep nagging Creality3D. They need to be reminded we are here.

sexycyborg, 2018-08-21:
I ask a couple of time a week, have no idea why they are making such a big deal of it...

sandro197802, 2018-08-22:
I don't understand too... This printer needs some basic Marlin feature such as MBL for its concave plate. I cannot recommend this printer to any other people that ask me, I seggest wanhao D6 or Monoprice maker ultimate because of the possibility to flash Marlin

sandro197802, 2018-08-22:
@thinkyhead Here in Italy we say "There is no worse deaf than those who do not want to hear"...

sandro197802, 2018-08-29:
I'm going ahead by myself 
[image]

sandro197802, 2018-08-29:
https://youtu.be/oZoFR2P4yrg

sandro197802, 2018-09-07:
Creality3d has finally released the source code!!!

closed 2018-09-07 by sandro197802