MarlinFirmware/Marlin #11430: M122 - TMC Debugging does not display data for my E0 driver (TMC2130).

HansJakob, 2018-08-01:
### Description

I have a DIY CoreXY printer with an MKS Gen 1.4 controller. All drivers are TMC2130 connected to SPI. I'm using hardware spi, but CS is not connected to standard pins. CS-X is pin 44, CS-Y is pin 40, CS-Z is pin 42 and CS-E0 is pin 59.

All drivers are configured correctly via the Marlin configuration, and they all work properly.

My problem is that I can not get data from my extruder driver when I send an M122 (TMC debug) command to the printer. I only receive data from X Y and Z. This did not happen in version 1.1.8. Here I received data from X Y Z and E0 as expected. I have not changed my hardware since I updated Marlin from 1.1.8 to 1.1.9.

If I send an M906 (TMC Motor Current), I also do not get data from E0, but only from X Y and Z.

However, if I send an M911 (TMC OT Pre-Warn Condition), I get data from all four drivers. Ie the M911 command works as expected.

### Steps to Reproduce

I updated from Marlin version 1.1.8 to version 1.1.9.

**Expected behavior:**
I expected the M122 and M906 commands to give me information about all four TMC2130 drivers mounted on my controller.

**Actual behavior:** 
I only receive data from the X Y and Z drivers and lack data from the E0 driver.

#### Additional Information

I have included my configuration files in a zip package. The files are:
Configuration.h
Configuration_adv.h
pins_RAMPS.h

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2250867/Marlin.zip)

teemuatlut, 2018-08-01:
1.1.9 broke a few things for configurable drivers and needs a hotfix patch.

teemuatlut, 2018-08-03:
https://github.com/teemuatlut/Marlin/tree/bf1_fix_smart_drivers

HansJakob, 2018-08-03:
This fix definitely resolved my problem. I can now see data for all four drivers.

Hywelmartin, 2018-08-05:
It worked for me as well. Thanks @teemuatlut for a quick fix

ddjikic, 2018-08-10:
@teemuatlut thank you it worked for me also

dmitry-kutergin, 2018-08-11:
Will it be integrated into a main repo?

thinkyhead, 2018-08-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. There are still some issues with `M122` but it should at least show all the axes now.

npmomchev, 2018-08-17:
My hardware config:
RuRAMPS4D v1.3 + ArdiunoDUE + 4 x TMC2130
software: PlatformIO (full updated) with Athom + Marlin bugfix-2.0.x (last)

M122
		X	Y	Z	E0
Enabled		false	false	false	false
Set current	1000	1000	1000	1000
RMS current	994	994	994	550
MAX current	1402	1402	1402	776
Run current	17/31	17/31	17/31	17/31
Hold current	8/31	8/31	8/31	8/31
CS actual		8/31	8/31	8/31	31/31
PWM scale		0	0	0	255
vsense		0=.325	0=.325	0=.325	1=.18
stealthChop	true	true	true	true
msteps		32	32	32	0
tstep		1048575	1048575	1048575	4294967295
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	false	false	false	true
OT prewarn has
been triggered	false	false	false	false
off time		5	5	5	15
blank time	24	24	24	54
hysteresis
-end		2	2	2	12
-start		3	3	3	8
Stallguard thrs	-10	-10	0	0
DRVSTATUS	X	Y	Z	E0
stallguard					X
sg_result		0	0	0	1023
fsactive					X
stst		X	X	X	X
olb					X
ola					X
s2gb					X
s2ga					X
otpw					X
ot					X
Driver registers:	X = 0x80:08:00:00
	Y = 0x80:08:00:00
	Z = 0x80:08:00:00
	E0 = 0xFF:FF:FF:FF

For Test I replaced the E0 TMC2130 with X1 TMC2130 - the result is the same.

closed 2018-08-18 by thinkyhead