MarlinFirmware/Marlin #11431: [Info] M600 inside gcode

GMagician, 2018-08-01:
I use cura as slicer and in gcode I have layer references. At beginning of layer 172 I wrote M600 (change filament).
Printer stopped at layer 174...is it normal?

ManuelMcLure, 2018-08-01:
Try inserting an `M400` gcode before the `M600` so the planer syncs up with the actual motion.

GMagician, 2018-08-01:
M400 just calls `planner.synchronize()` and `pause_print` also calls it (it's called by M600) so I think it will not solve, but I'll try

thinkyhead, 2018-08-01:
It should execute `M600` at the point it appears in the G-code. How are you reckoning the layers? Note that `M600` will raise Z by a couple of millimeters as part of pausing the print.

GMagician, 2018-08-02:
I'm almost sure about layer position, I've done a double check:
1) Cura says 176 total layers. I print with 0,2mm layer height. Piece will close screw hole 1mm before end of piece. This will give me 5 layers to close it. 176-5 = 171 then 172 is where it start to close.
2) Cure let you see layer by layer what happens. at layer 172 I see it close the hole.

I had to pause manually by LCD when I saw it was closing hole without stopping. Then I inserted screw and restarted print (really I need only a pause but even M125 doesn't work). After restaring pring. Printer stopped some layer after 172 (I saw 174 on repetier...yes I use cura for slice and repetier as host :-D)

GMagician, 2018-08-02:
Ok Cura starts from layer 0 and I have to stop it when it says layer 170...

closed 2018-08-02 by GMagician