MarlinFirmware/Marlin #11432: [2.0.x] FWRetract moves Z axis when Hop is set to 0 and UBL is enabled

tcm0116, 2018-08-02:
### Description

In bugfix-2.0.x (downloaded today), sending G10/G11 with the Z-Hop value set to 0 in the LCD causes the Z axis to move a small amount in the positive direction with each command. These small amounts eventually add up and the nozzle slowly drifts away from the bed.

### Steps to Reproduce

1. Compile bugfix-2.0.x with firmware retraction and UBL enabled.
2. Set the Z-Hop value to 0.
3. Build mesh and enable UBL.
4. Send G10 and G11 commands.
5. Watch the z axis move slightly with each command.

**Expected behavior:** The Z-axis shouldn't move.

**Actual behavior:** The Z-axis moves.

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2251723/Configuration.zip)

tcm0116, 2018-08-02:
@Roxy-3D any ideas on this?

tcm0116, 2018-08-07:
I've been looking into this, and I think this might be occuring due to a an issue in `SYNC_PLAN_POSITION_KINEMATIC()`, which is called in `FWRetract::retract()`.

I uncommented the debug serial echos in `FWRetract::retract()` and added some additional logging of the stepper positions. Below is the output that I receive when running `G11` for the first time after a fresh reboot and home of the machine. The issue is that `stepper.position[z]` is changing from 4508 to 4516. The Z steps/mm on my machine is 400, so what's happening is that when the stepper position is re-computed in `sync_plan_position()`, the new Z position of 4516 is reasonable when multiplying 11.29 by 400.

```
Send: G11
Recv: retracting 0
Recv: swapping 0
Recv: active extruder 0
Recv: retracted[0] 1
Recv: retracted_swap[0] 0
Recv: retracted[1] 0
Recv: retracted_swap[1] 0
Recv: current_position[z] 11.29
Recv: current_position[e] 0.00
Recv: stepper.position[z] 4508
Recv: stepper.position[e] 0
Recv: hop_amount 0.00
Recv: echo: cold extrusion prevented
Recv: echo: cold extrusion prevented
Recv: retracting 0
Recv: swapping 0
Recv: active_extruder 0
Recv: retracted[0] 0
Recv: retracted_swap[0] 0
Recv: retracted[1] 0
Recv: retracted_swap[1] 0
Recv: current_position[z] 11.29
Recv: current_position[e] 0.00
Recv: stepper.position[z] 4516
Recv: stepper.position[e] 0
Recv: hop_amount 0.00
Recv: ok
```
After thinking about it some more, it occurred to me that the offset must be due to the UBL leveling correction. To prove this, I moved to a mesh grid point (10, 10), and ran G11. Here's my topology report:

```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,260)              (210,260)
Recv:         0       1       2       3
Recv:  4 | +0.295  +0.202  -0.050  -0.260
Recv:    |
Recv:  3 | +0.172  +0.055  -0.145  -0.240
Recv:    |
Recv:  2 | +0.100  +0.060  -0.045  -0.080
Recv:    |
Recv:  1 | +0.097  +0.042  +0.080  +0.040
Recv:    |
Recv:  0 |[+0.090] +0.142  +0.127  +0.240
Recv:         0       1       2       3
Recv:     ( 10, 10)              (210, 10)
Recv: 
Recv: ok
```

And here's the output of G11:

```
Send: G10
Recv: retracting 1
Recv: swapping 0
Recv: active extruder 0
Recv: retracted[0] 0
Recv: retracted_swap[0] 0
Recv: retracted[1] 0
Recv: retracted_swap[1] 0
Recv: current_position[z] 11.29
Recv: current_position[e] 0.00
Recv: stepper.position[z] 4552
Recv: stepper.position[e] 0
Recv: hop_amount 0.00
Recv: echo: cold extrusion prevented
Recv: echo: cold extrusion prevented
Recv: retracting 1
Recv: swapping 0
Recv: active_extruder 0
Recv: retracted[0] 1
Recv: retracted_swap[0] 0
Recv: retracted[1] 0
Recv: retracted_swap[1] 0
Recv: current_position[z] 11.29
Recv: current_position[e] 0.00
Recv: stepper.position[z] 4516
Recv: stepper.position[e] 0
Recv: hop_amount 0.00
Recv: ok
```
The current Z position is still 11.29, but the stepper position at the beginning of `FWRetract::retract()` is now 4552, but still goes back to 4516 at the end. If we divide 4552 by 400, the result is 11.38, which is 0.09 greater than the current Z position of 11.29, which corresponds with the mesh offset at (10, 10).

Based on the above, I believe that `sync_plan_position()` needs to be updated to take the UBL offset into account. It may need to take the offset from other ABL systems, but I haven't looked into this as of yet.

Roxy-3D, 2018-08-07:
> Based on the above, I believe that sync_plan_position() needs to be updated to take the UBL offset into account. It may need to take the offset from other ABL systems, but I haven't looked into this as of yet.

Yes.  That is very possible.    When UBL was written, it followed what the original Mesh Bed Leveling did.   The original Mesh Bed Leveling has a Motion.cpp file.   And when that was called, any movement was broken down into segments that crossed mesh boundary lines.   It called itself recursively.   And where ever the nozzle currently was located didn't matter.   It was the destination that mattered.   The destination of the line segment would have the Z-Height correction calculated and added onto the Z, and then the planner was called with the new line segment.

UBL is similar, except the UBL_motion.cpp is optimized for CPU calculations.  And the recurrsion is removed.   But the rest is similar.    Every move is broken down to smaller segments that cross mesh lines.   And the Z-Height correction is added on to the Z at the end of those segment moves.

But can you explain what Retraction has to do with Z-Correction????   Maybe that is at the root of the issues we are seeing?

tcm0116, 2018-08-07:
> But can you explain what Retraction has to do with Z-Correction????

That what pretty much what I was trying to do with my previous post. However, to simplify, they become related by the fact that Firmware Retraction moves the Z axis, and then spoofs the position such that the reported position is not changed. This is accomplished by actually moving the Z axis, then setting `current_position[Z_AXIS]` back to what it was, and then calling `SYNC_PLAN_POSITION_KINEMATIC()`.

In reality, it's the call to `SYNC_PLAN_POSITION_KINEMATIC()` that is causing the issue, so any other code that calls that family of functions while UBL is enabled is likely to cause the actual Z position to become offset by the UBL correction that that point.

I just tried offsetting the actual stepper position in `Planner::_set_position_mm()` by the value returned by `ubl.get_z_correction()`, but it did not resolve the issue. The reported Z stepper position in G10 and G11 looked better, but it still moved after the call to G11 (which is what it was doing before). On the next call to G10, the reported Z stepper position was then offset, so it's at least tracking with what's happening, but I need to find where that's occurring.

Roxy-3D, 2018-08-07:
In the 'Big Picture'...   The 'Right' answer is to move the UBL Motion routine down into the Planner.    The problem with doing that is we would lose the very CPU optimized code for generating line segments.

I eagerly await your recommendations!!!!

brucehvn, 2018-09-09:
I just posted in #11512 but then saw this issue.  This seems to be exactly what I'm seeing, except my z-axis moves in the negative direction on retract/unretract.  I have zhop set to 0 and I'm using UBL.

tcm0116, 2018-09-09:
@brucehvn can you post your configuration files along with a gcode file that exhibits the issue?

In #11486, you mentioned that it happens with both auto and manual retraction. Are you referring to firmware retraction in both cases (i.e. enabling/disabling auto retraction in the retraction menu on the LCD), or do you mean firmware retraction for auto and slicer retraction for manual?

boelle, 2019-03-06:
@tcm0116 have you retested this with todays copy of 2.0?

closed 2019-03-07 by tcm0116