MarlinFirmware/Marlin #11433: BLTouch in 1.1.9

oziee, 2018-08-02:
### Description
latest Marlin 1.1.9 using Ardunio 1.8.5

compiling after enabling the BLTouch i get this error

`
sketch/Marlin_main.cpp: In function 'void servo_init()':
Marlin_main.cpp:950: error: 'STOW_Z_SERVO' was not declared in this scope
     STOW_Z_SERVO();
                  ^
sketch/Marlin_main.cpp: In function 'void bltouch_command(int)':
Marlin_main.cpp:2035: error: 'MOVE_SERVO' was not declared in this scope
       MOVE_SERVO(Z_PROBE_SERVO_NR, angle);  // Give the BL-Touch the command and wait
                                         ^
exit status 1
'STOW_Z_SERVO' was not declared in this scope
`

### Steps to Reproduce

set the board to the Sanguino 1284P 16MHz for the Creality Ender 3

#### Additional Information
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2252006/configs.zip)

Thanks

Roxy-3D, 2018-08-02:
@thinkyhead    Is bugfix_1.1.x frozen?  Or can we still make changes to it?    I know we don't see things the same way on this, but I think even 1.1.x should have known good, low risk bug patches applied to it.   But no new features.

Can you explain your vision?   Because there are a couple of IDEX fixes that SilentNinja is working on and depending upon how we handle the freeze, it affects what work should be done.   Personally, I don't know that 1.1.x needs much in the way of IDEX.  It might be ok to focus all that work on 2.0.0.  

But right now...   Even I am am confused about the direction.    Can you give a few lines of guidance?   And I'm hoping very low risk...  Very simple patches that fix bugs but add no new features can still be applied to both 1.1.x and bugfix_1.1.x.

elfag, 2018-08-03:
When i uncomment safe homing, i got it to work

sglbt, 2018-08-29:
Hi everyone,

I just try to compile latest Marlin 1.1.9 for my creality CR10 and I got the same error, even after enabling safe_homing...
Can anyone help me ?

Thanks in advance for your help.

thinkyhead, 2018-08-30:
You need to define the servo pins you're using for your setup. Do you know what the servo pins should be for the Creality Melzi board? If they are standardized, then we can add them to the pins file. If they are ad hoc, then just define `SERVO0_PIN` yourself in your `Configuration.h`. In the future we'll provide a better error message.

sglbt, 2018-09-03:
Hi thinkyhead,

Sorry. I was very busy in the pas week.
I would like to thank you for your answer.
I added "SERVO0_PIN 11" (for my CR 10) to my configuration.h file and everything work properly.

Thnak you again.
sglbt

sglbt, 2018-09-10:
Sorry, for those you are playong with a CR10, I made a mistake, it's not pin 11 but pin 27.
Here is what have to be added just after #define BLTOUCH

#define SERVO0_PIN 27

Everything works perfectly now.
Sglbt

closed 2018-11-02 by thinkyhead

JoBo1943, 2018-11-14:
Sglbt. I'm having the same error on my ender 3 that uses the same CR10 control board and have already defined SERVO0_PIN 29, a unused pin on my board.
Could you please post your configuration.h file?

thanks Joe