MarlinFirmware/Marlin #11434: Multiple Driver and SPI doesnt work.

sinus23, 2018-08-02:
### Description

Mixed / Multi Driver and SPI dont work
But in configure_adv.h, if you use the 
MINMUM_STEPPER_DEALY 650
MINIMUM_STEPPER_PULSE 2
MAXIMUM_STEPPER_RATE 250000
it seems to be fixed. 

### Steps to Reproduce
if you used TMC2130 on X & Y with SPI and other DRV8825 works fine
if you used TMC2130 on X&Y&Z, DRV8825 on E0 dont work

1.1.8 works fine with mixed Driver

The driver on E0 still consumes power, E0 stepper motor does not hum but no movement, with more amps also not, stepper manufacturer exchange brings also nothing

JohnnyTheOne, 2018-08-02:
Try
MAXIMUM_STEPPER_RATE 400000

sinus23, 2018-08-02:
Nope dont work, only 250000 working there. 
#define MAXIMUM_STEPPER_RATE 250000
#define MINIMUM_STEPPER_PULSE 2
#define MINIMUM_STEPPER_DIR_DELAY 650
But this actually reduces the TMC

dsmal150, 2018-08-03:
Hi, I have a similar problem. Mixed extruder XYZ TMC2130 and E0,E1 DDRV 8825. When i change steps per/mm from 280 to 716 seems that works, but little moves not correctly, seems like a some steps are missing.

sinus23, 2018-08-03:
dsmal150 conf adv:
#define MAXIMUM_STEPPER_RATE 250000
#define MINIMUM_STEPPER_PULSE 2
#define MINIMUM_STEPPER_DIR_DELAY 650
that works just fine every time, but no longer the full potential of the TMC, no missing steps

dsmal150, 2018-08-06:
sinus23 maybe try, using repair drivers.h file and commenting not used drivers in configuration.h Momenttly i Have XYZ and E0 at TMC2130 but E1 function in drv 8825 normal.

closed 2018-11-05 by thinkyhead