MarlinFirmware/Marlin #11435: BLTouch not probing Z_MIN / issue on Geetech configs

FrancYescO, 2018-08-02:
Hello i have a Geetech Pro W with a second Extruder mount + BLTouch, i wanted to try the marlin firmware and i just downloaded the 1.1.9 build, than basically to have all working i merged configs from:

https://github.com/MarlinFirmware/Marlin/tree/1.1.x/Marlin/example_configurations/Geeetech

Taken the Pro B probe config to have the BLTouch working, and the ProC/W to for the basic and dual extrusion, my board is a BOARD_GT2560_REV_A_PLUS.

**The final issue** is that i cannot Home Z: the probe is working on selftest but when homing Z will not be activated, the pin stay retracted, the probe/extruding block will go in the center of bed and go down until crashes into bed

Can someone help me to find out whats wrong in my config? probably the USE_ZMIN_PLUG / USE_ZMAX_PLUG

Another thing: i saw that in the Pro B example configs there are some value about the `Geeetech MK8 Extruder`, `MK2a PCB Heatbed` and `Bed Skew Compensation` as ProB/C/W have similar hardware should these config be copied to the others (ProC/W) example config?

Seref, 2018-08-02:
Same happens in Marlin 2.x bugfix.

When starting the Printer it (mostly) works fine, auto homing works and it stops when MIN_Z is triggered.

After I finished a print and start another one MIN_Z triggers but it ignores it for some reason until I manually auto home it again over the menu or restart the printer.

FrancYescO, 2018-08-02:
I was pretty sure was a configuration issue for me infact i solved the homing issue enabling 
`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`
and removing 
`#define Z_MIN_PROBE_ENDSTOP` and pin definition

Now i have a worst issue:

The nozzle is not able to reach, or will go with much "difficulty" to 200c, will go fast until 195 than go down, and again up to 200 but is not stable ( probably is due to the changed `DEFAULT_Kp/i/d`?)... then will start the print, but the nozzle is not moving and the completion % will go fast to 100% without printing nothing ... can someone point me out to what i have to look in the config?

GMagician, 2018-08-02:
BLtouch **is not used** to home machine if Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is disabled.
In such situation it is used only for bed leveling.

Since BOARD_GT2560_REV_A_PLUS has Z_MIN connected to pin 30 there are 2 cases:
  1) this input is not used and then you have only to connect bltouch there
  2) this input is used then bltouch is supposed to be used only for bed leveling

I don't know anything about such configuration then I can't say more on this

nacho87, 2018-08-02:
@FrancYescO I comment this lines 

//#define PROBING_HEATERS_OFF       // Turn heaters off when probing
//#define PROBING_FANS_OFF          // Turn fans off when probing

elfag, 2018-08-02:
I have the same problem

GMagician, 2018-08-02:
@FrancYescO please don't post different issues in the same thread. Open a new one to avoid misunderstandigs.

@elfag What problem do you have?

elfag, 2018-08-03:
@FrancYescO  that bltouch dont work as zmin

FrancYescO, 2018-08-03:
I solved my issue as i've wrote in the 3rd comment

Changing kd/I/k also solved the temperature issue

closed 2018-08-03 by FrancYescO