MarlinFirmware/Marlin #11436: Z_Min gets ignored after a finished print (bugfix-2.0.x)

Seref, 2018-08-02:
When starting the Printer it (mostly) works fine, homing (G28 in the gcode not over the menu) works and it stops when MIN_Z is triggered.

After I finished a print and start another one MIN_Z triggers but it ignores it for some reason until I manually home it again over the menu or restart the printer.

Im using an kfb2.0 board which is basically an Atmega2560 and a Ramps 1.4 EFB board and an inductive probe (which also lights up and I rechecked the cables they are all working).

GMagician, 2018-08-02:
Limit switches are checked only while homing. If you need a constant check you have to enable ENDSTOPS_ALWAYS_ON_DEFAULT or play with M120, M121 in your gcode

Seref, 2018-08-02:
@GMagician I don't need a constant check, after I finished printing I printed something else and during that homing sequence the MIN_Z triggering didnt stop the Z Axis. Because of that the hotend rammed the heatbed, this happened two times.

GMagician, 2018-08-02:
ops sorry, misunderstood

GMagician, 2018-08-02:
Thid is what I don't undestood:

> After I finished a print and start another one MIN_Z triggers but it ignores it for some reason until I manually home it again over the menu or restart the printer.

So you are saying that during second print (where you have in gcode G28) it crash on the bed while homing?

closed 2018-08-04 by Seref