MarlinFirmware/Marlin #11437: Marlin 1.1.9 & marlin 1.4 & LCD 2004 buzzer not working

stoicastefanv, 2018-08-02:
Hi all i had tried to upgrade the firmware from 1.1.8 to 1.1.9 and have an issue regarding the buzzer when using the menu with LCD 
using Marlin 1.1.9
i head enabled 
```

#define MOTHERBOARD BOARD_RAMPS_14_EFB
#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define SPEAKER
```

**with above settings Speaker not working** 
with Marlin 1.1.9 & Pronterface 
`M300 S4000 P500`
speaker is working

with Marlin 1.1.8 & Pronterface 
`M300 S4000 P500`
speaker is working

I had tried to look in pins_ramps to see if something had changed but is little too much for me

boelle, 2018-08-03:
you dont need speaker.... speaker is real speaker and not a buzzer

stoicastefanv, 2018-08-06:
Maye was an upload error, today i had uploaded the firmware and guess what? it works !! sorry for this post , now is closed

closed 2018-08-06 by stoicastefanv