MarlinFirmware/Marlin #11441: Official 1.1.9 - "START_BMPHIGH" shift logo down (Graphical LCD 12864)

brunosso, 2018-08-03:
### Description

"START_BMPHIGH" start to draw logo not at the top of LCD screen
### Steps to Reproduce

1. uncomment "START_BMPHIGH"
2. logo is offsetted down

**Expected behavior:** Logo start at the top of LCD

**Actual behavior:** logo is offsetted down

#### Additional Information

I think the problem is the variable offy in the ultralcd_impl_DOGM.h file. A workaround for now is:

```
constexpr uint8_t offy =
      #if ENABLED(START_BMPHIGH)
        //(LCD_PIXEL_HEIGHT - (START_BMPHEIGHT)) / 2
        0
      #else
        DOG_CHAR_HEIGHT
      #endif
    ;
```

AnHardt, 2018-08-03:
The current code tries to center the bitmap - not starting at the top. This is intentional - not a bug. If you want it to be different change it for your Marlin to fit your needs.
Please don't alter/replace the Marlin logo. Please add your custom logo. It's a matter of honoring Marlin.

brunosso, 2018-08-03:
OK thanks for the reply. With this behaviour the string line of version and website will be overwrited...
I want to display the Marlin logo, i have already setup a custom bootscreen