MarlinFirmware/Marlin #11442: [1.1.9] Improvement : TMC2130 add LCD menu for StealthChop or Spreadcycle mode

pipersw, 2018-08-03:
Could you add a menu in LCD in case of use TMC2130 for switching between StealthChop or Spreadcycle mode ? it could be usefull for testing purpose or if we would be quiet at night and powerfull the day without the need of re compiling the code

Atatoth, 2018-08-03:
Good idea :)

oakdesign, 2018-08-03:
If you TMC are configured in hybrid mode you already have this Basic function on Hand just with M913 set the threshhold Speed to a value that the Driver will not Switch and stay in the desired mode

teemuatlut, 2018-08-03:
It's coded and coming. No timetable though at this point as there are quite a few items on the list before this one.

Seref, 2018-08-05:
Would this also be possible on TMC2208?

Flawl3ssSWE, 2018-08-05:
Seref, I don't think so since the TMC2130 can be controlled from the firmware (SPI) and is there able to set the different modes, but the TMC2208 is configured thru UART and not thru the firmware.

Seref, 2018-08-05:
@Flawl3ssSWE but according to the spec sheets you can even though it's over UART and not SPI controll the driver and set the modes over that. And Marlin also offers an option to use the UART connection of the TMC2208.

oakdesign, 2018-08-06:
[img]https://up.picr.de/33495745bo.jpg[/img]
2208 Wired die UART Firmware comm

boelle, 2019-03-13:
@pipersw still with us?

pipersw, 2019-03-14:
Yes

sjasonsmith, 2019-11-08:
@boelle, I think this is done and can be closed? I used that menu today!

I can't confirm that it works with TMC2130 drivers, but it was there with my 2208s!

closed 2019-11-08 by pipersw