MarlinFirmware/Marlin #11459: [1.1.9]

Kazza2003, 2018-08-05:
### Description

I recently updated my printer setup (Hypercube Evolution - Core X/Y, powered by a RAMPS 1.6 + DRV8825) to 1.1.9. I copied all the settings from my old 1.1.8 config and config_adv over and adjusted them for 1.1.9. Now I have the problem that the extruder (e3d Titan) makes loud noises if it retracts (grinding noise). I did not change anything at the hardware prior or after the firmware update to 1.1.9. Can it be that marlin 1.1.9 handles or calculates retraction different?

### Steps to Reproduce

1. [First Step]
Upload Marlin 1.1.9 and copy all existing settings from your old config/condig_adv
2. [Second Step]
Flash and test print

**Expected behavior:** 
A hearable but still silent retraction sound (fast moving gears).

**Actual behavior:**
A loud, grinding noise like it is overreacting.

#### Additional Information

Zip of my configs is attached.

gloomyandy, 2018-08-05:
There does not seem to be any file attached. Do you have the type of stepper configured correctly (this is a new config in 1.1.9) search for X_DRIVER_TYPE in configuration.h, you need to set this to whatever drivers you have (DRV8825?).

Kazza2003, 2018-08-05:
Yes, I already set them up for all my motors. They are properly set to DV8825. That is what confuses me.

gloomyandy, 2018-08-06:
I think if you want further help you need to upload your configuration files so folks can check to see if there is anything odd in there. Also did you enable any of the new 1.1.9 features (like S curve acceleration or junction deviation)? You may also want to try slowing your extruder acceleration/jerk and retract speeds as 1.1.9 probably implements them more accurately and your old values may not work so well.

Kazza2003, 2018-08-06:
I did attach them in a .rar file because the page didn't want to accept my config files in raw form. But I can add it again.

Kazza2003, 2018-08-06:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2262832/Config.zip)

ismaweb, 2018-08-12:
Same problem here, with DRV8825 also
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2280896/Configuration.zip)

ismaweb, 2018-08-12:
Video showing the noise and the extruder assembly vibrating:
[rattling.mp4.zip](https://github.com/MarlinFirmware/Marlin/files/2280901/rattling.mp4.zip)

Kazza2003, 2018-08-12:
Interesting. X and Y are super silent (relatively silent for a DRV8825) since 1.1.9 but the extruder is totally out of his mind.

Kazza2003, 2018-08-12:
Gonna check your config for a second ismaweb and see if there are any differences in ours. Maybe we can find the cause ^^.

Kazza2003, 2018-08-12:
For starters you did not assign the DRV8825 to your extruder. So no wonder it might not work properly. Also doesn't have your Printer 2 Z motors, moved by 2 different Z Stepper Motors?

ismaweb, 2018-08-12:
@Kazza2003 My board is a GT2560 which has a single controller to drive two tween Z motors.

@lrpirlet I have DRV8825 for XYZ and A4988 for E0, I think my Configuration.h is correct. Isn't it?

I´ve just swaped (the phisical controllers) X, Y and Z DRV8825 to A4988 and cofigured properly in Configuration.h:
Exact same behavior, smooth travels, and "teletype" noise while extruding.

ismaweb, 2018-08-12:
After few hours I managed to get I work smoothly:
My software, in some circumstances, not allways, was recording a 0 acceleration on the eeprom
(gcode M204 P0)
If I manually send a M204 P400, it runs like a charm.

So, @Kazza2003 check your eeprom values trough manual gcode M501 and look for weird values

DKDarkness, 2018-08-12:
I am having the same issues with retract/prime on my cartesian style printer and DRV8825 and titan extruder. Normal extrudes work fine, but when retracting/priming after retract it just makes a grinding noise and does not retract/prime. Changing between DRV8825 and A4988 setting in configuration.h does no difference. Disabling S acceleration or setting M204 P400 manually makes no difference.

Kazza2003, 2018-08-12:
@lrpirlet gonna try that first now with a test print.

ismaweb, 2018-08-13:
@Kazza2003, sorry, look at the output from the M503 comand which are the actual used values, M501 are the eeprom ones wich are superceded with M503 at execution time:
Find the M203 values, E value is your max extruder speed (retract also)
Then find the M204 values, R is retract acceleration.

Kazza2003, 2018-08-13:
M201 X2000 Y2000 Z100 E1000
M204 P1000.00 R1000.00 T1000.00

 I think you mistyped @ismaweb. It is M201 you wanted to know or?

ismaweb, 2018-08-13:
@Kazza2003, I want M203 (Maximum feedrates). I'm suspecting that you have a massive speed/acceleration configured that makes a step loose on retractions.

Other thing you can look at:
Have you tuned the amperage in the driver (the tiny pot)? 
Too low => loosing steps => noise
Too high => noise

ismaweb, 2018-08-13:
Ok @Kazza2003, everything looks normal

mgaller, 2018-08-19:
I have the exact same problem on my Tevo Tornado with TMC2130 on X,Y,Z and LV8729 on E0.
Worked fine with 1.1.8 before, tried several things, nothing worked.

gloomyandy, 2018-08-19:
@mgaller  can you upload your configuration files and give details of what you have tried so far.

mgaller, 2018-08-19:
The LV8729 is running in 1/32 mode with 0.9 Volt vref.
The extruder is not moving at all, just a weird noise.

[Marlin-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/2300297/Marlin-1.1.x.zip)
So what I tried so far:
- changed MINIMUM_STEPPER_DIR_DELAY and the other two vaues for LV8729
- disabled TMC2130
In the Tevo Tornado facebook group I found a Marlin 1.1.9 with TMC2130 and LV8729.
Uploaded it, now the extruder is moving but still some noise.
Then I noticed that the steps were half my steps, so I changed the jumpers to 1/16.
Then, it was the same like before, weird noise no movement.
 I upped the vref for the LV8729 to 1.1V. Then I got movement again, but still weird noise.
Changed back to 1/32 and used again the doubled steps, only weird noise.
Tried now for like 4 hours, I give up now.
I think something is messed up with the driver control.

mgaller, 2018-08-19:
So I played a little more around, the problem seems related to MAXIMUM_STEPPER_RATE.
I if set it to 130000 "Maximum for LV8729 stepper driver" it is not working.
But I if lower it to 15000 then the extruder is working fine.
Still using LV8729 1/32 with 0.99V

thinkyhead, 2018-08-21:
@ejtagle — What do you make of this? ☝️

ejtagle, 2018-08-22:
DRV8825 requires to increase a bit the pulse width (i still have to port the timing fixes for that!) ... The retract issue is probably caused by Linear advance that does not perfectly respect maximum speed.. This is something that is very hard to fix (LA requires to fast-forward movements in order to "advance" pressure...

DKDarkness, 2018-09-01:
Tried changing the maximum feedrates with no effect. Ended up lowering the retraction speed to 20mm/s in cura 3.4.1, which seems to be about the maximum it can do. Running DRV8825@28v, 0.7A on E3D pancake stepper (0.9 degree) and Titan extruder with 3:1 ratio. Marlin 1.1.9.

thinkyhead, 2018-09-05:
Halving the number of micro-steps for E may help.

AllWashedOut, 2018-10-17:
I had issues similar to this after enabling linear advance in marlin 1.1.9 - extruder would skip steps a few times per model. Applying the patch from pull/11759 resolved the issue.
https://github.com/MarlinFirmware/Marlin/pull/11759

boelle, 2019-02-20:
@Kazza2003 is this still an issue when using latest bugfix 2.0?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead

subtyper, 2019-05-12:
Hey guys,

I have just updated my marlin firmware to 1.1.9 and added a bltouch configured using teaching techs YouTube video am also using octoprint on a raspberry pi 3b+ my issue is when I move the z stepper either up or down from menu it just moves the stepper down and also the motor sounds like a teletyper really loud. Anything possible I could do to fix this