MarlinFirmware/Marlin #11464: IDEA: Extract selftest from prusa??

boelle, 2018-08-05:
hi

just wondering  if it would make sense to extract the selftest procedure from here?

https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/ultralcd.cpp

of course parts like testing the fans requires the tacho signal to be feed back

throw in your 0.02$ :-D

forkoz, 2018-08-09:
Prusa has the advantage of uniform hardware. My fans don't have speed sense. I'd have to downgrade the speed of the fans or get larger ones. It can't test my end stops but homing already does that. Power to the HE/bed is tested by heating it up, if it doesn't heat for long enough the printer shuts off, etc. 

So in short I don't there would be much of a point.

Grogyan, 2018-08-09:
@boelle Josef's Power Panic is seriously simple, and because it is designed for being on the back of a IEC connector, means that it should already be mass made.

The self test can be put into Marlin, however, because a lot of people use a lot of different electronics, it is hard to know if there are any available pins to read tachos.

Once Marlin is 32bit ready, this can be something to look at again I reckon.  As 32bit micro's have a lot more pins available than 8bit AVRs

boelle, 2018-08-09:
with a Ramps board there is plenty free pins on aux1 or aux2

of course a small addon board is needed as tacho can not go directly to the MCU nor can thermistors or the voltage monitor input needed

forkoz, 2018-08-10:
Yea, I struggled to find interruptible pins for end stops. Add somebody using TMCs into the mix and all those self tests are getting //

Roxy-3D, 2018-08-10:
I understand the desire to have interruptible pins for the end stops...  But...   The 8-Bit AVR's at 20 MHz work just fine with Programmed IO on the end stops.   A 32-bit processor at 100 MHz really doesn't need interrupts on its end stops.

With that said...  I do understand people want certain things even if they aren't necessary!!!

boelle, 2018-08-10:
my idea was to have some kind of self test as an option, not as a forced thing

and if prusa can do it so can marlin, and their ensy board is nothing more than a glorified ramps board with some extra flash memory. i did look for quite a bit on the schematic that ultimachine has published and did not find much new there other than that the power panic also disables the heaters, but only if it gets a signal (if left unconnected it will not disable anything)

since i have a slow day and we have a pause in those heatwaves here in europe i will look at it again and see what changes are really needed on a Ramps and also on a Ramps FD v2.3, if nothing else just for the fun of it and to keep my mind in gear :-D

boelle, 2018-08-10:
@Grogyan so no one is producing these power panic boards?

closed 2018-10-22 by boelle

thinkyhead, 2018-10-26:
Geeetech's current crop of machines includes early power-loss detection and a big capacitor to hold the charge for several seconds.