MarlinFirmware/Marlin #11468: MARLIN 1.1.9 ERROR COMPILING USING BOARD_MKS_15

LivioB74, 2018-08-05:
HI !
I have download new firmware ... after set my params and try compiling ... error !
To find where i have wrong i will try compiling every new mod ... and i start ...
So i have used a fresh download, in configuration.h i have only changed  BOARD_RAMPS_14_EFB with BOARD_MKS_BASE_15... Verify ... and error !!
I think some pin declaration conflict, but i'm not so skilled to be sure ... if you want verify ... thanks again in advance, and i'm sorry for my poor english. I hope you have understand ...
What i can do to use Marlin 1.1.9 (stable) just released with my mks base 1.5 ?
If you try test ... but i think result will be 
This error msg ... 

Arduino:1.9.0-beta (Mac OS X), Scheda:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/HAL.h:33:0,
                 from sketch/MarlinConfig.h:39,
                 from sketch/Marlin.h:35,
                 from sketch/stepper.cpp:80:
pins_MKS_BASE_15.h:32:34: error: pasting "/* Digital 3  / Pin 5   / PE3  / SERVO2_PIN*/" and "_DDR" does not give a valid preprocessing token
 #define X_MS1_PIN            5   // Digital 3  / Pin 5   / PE3  / SERVO2_PIN
                                  ^
sketch/fastio.h:87:42: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                          ^
sketch/stepper.cpp:2387:5: note: in expansion of macro 'SET_OUTPUT'
     SET_OUTPUT(X_MS1_PIN);
     ^
sketch/stepper.cpp:2387:16: note: in expansion of macro 'X_MS1_PIN'
     SET_OUTPUT(X_MS1_PIN);
                ^
pins_MKS_BASE_15.h:32:34: error: pasting "/* Digital 3  / Pin 5   / PE3  / SERVO2_PIN*/" and "_PIN" does not give a valid preprocessing token
 #define X_MS1_PIN            5   // Digital 3  / Pin 5   / PE3  / SERVO2_PIN
                                  ^
sketch/fastio.h:87:61: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                                             ^
sketch/stepper.cpp:2387:5: note: in expansion of macro 'SET_OUTPUT'
     SET_OUTPUT(X_MS1_PIN);
     ^
sketch/stepper.cpp:2387:16: note: in expansion of macro 'X_MS1_PIN'
     SET_OUTPUT(X_MS1_PIN);
                ^
pins_MKS_BASE_15.h:33:34: error: pasting "/* Digital 6  / Pin 14  / PH3  / SERVO1_PIN*/" and "_DDR" does not give a valid preprocessing token
 #define X_MS2_PIN            6   // Digital 6  / Pin 14  / PH3  / SERVO1_PIN
                                  ^
sketch/fastio.h:87:42: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                          ^
sketch/stepper.cpp:2388:5: note: in expansion of macro 'SET_OUTPUT'
     SET_OUTPUT(X_MS2_PIN);
     ^
sketch/stepper.cpp:2388:16: note: in expansion of macro 'X_MS2_PIN'
     SET_OUTPUT(X_MS2_PIN);
                ^
pins_MKS_BASE_15.h:33:34: error: pasting "/* Digital 6  / Pin 14  / PH3  / SERVO1_PIN*/" and "_PIN" does not give a valid preprocessing token
 #define X_MS2_PIN            6   // Digital 6  / Pin 14  / PH3  / SERVO1_PIN
                                  ^
sketch/fastio.h:87:61: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                                             ^
sketch/stepper.cpp:2388:5: note: in expansion of macro 'SET_OUTPUT'
     SET_OUTPUT(X_MS2_PIN);
     ^
sketch/stepper.cpp:2388:16: note: in expansion of macro 'X_MS2_PIN'
     SET_OUTPUT(X_MS2_PIN);
                ^
pins_MKS_BASE_15.h:34:34: error: pasting "/* Analog 5   / Pin 92  / PF5*/" and "_DDR" does not give a valid preprocessing token
 #define Y_MS1_PIN           59   // Analog 5   / Pin 92  / PF5
                                  ^
sketch/fastio.h:87:42: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                          ^
sketch/stepper.cpp:2390:7: note: in expansion of macro 'SET_OUTPUT'
[... 1580 lines cut ...]
                                           ^
sketch/stepper.cpp:2511:25: note: in expansion of macro 'READ'
       SERIAL_PROTOCOLLN(READ(E1_MS2_PIN));
                         ^
sketch/stepper.cpp:2511:30: note: in expansion of macro 'E1_MS2_PIN'
       SERIAL_PROTOCOLLN(READ(E1_MS2_PIN));
                              ^
pins_MKS_BASE_15.h:41:34: error: pasting "/* Digital 4  / Pin 1   / PG5  / SERVO3_PIN*/" and "_PIN" does not give a valid preprocessing token
 #define E1_MS2_PIN           4   // Digital 4  / Pin 1   / PG5  / SERVO3_PIN
                                  ^
sketch/fastio.h:69:64: note: in definition of macro '_READ'
 #define _READ(IO)             TEST(DIO ## IO ## _RPORT, DIO ## IO ## _PIN)
                                                                ^
sketch/stepper.cpp:2511:25: note: in expansion of macro 'READ'
       SERIAL_PROTOCOLLN(READ(E1_MS2_PIN));
                         ^
sketch/stepper.cpp:2511:30: note: in expansion of macro 'E1_MS2_PIN'
       SERIAL_PROTOCOLLN(READ(E1_MS2_PIN));
                              ^
exit status 1
pasting "/* Digital 3  / Pin 5   / PE3  / SERVO2_PIN*/" and "_DDR" does not give a valid preprocessing token

Questo report potrebbe essere più ricco di informazioni abilitando l'opzione
"Mostra un output dettagliato durante la compilazione"
in "File -> Impostazioni"

AnHardt, 2018-08-05:
Arduino:1.9.0-beta has currently (sometimes?) problems with comments behind #defines.
Either use an earlier version of Arduino or remove the comments where you get problems.

LivioB74, 2018-08-06:
Thanks a lot ! I have tried with arduino 1.8.4 and Works great !!
In the next days i will try it on my printer ...

closed 2018-08-06 by LivioB74