MarlinFirmware/Marlin #11469: Bed PID Autotuning failure

JimF42, 2018-08-05:
### Description

I have recently upgraded my firmware from whatever version ANET was using for their A6 printer to 1.1.8 with modifications to support auto bed leveling.

Under the old FW I was able to run `M303 E-1 S50` just fine and got good PID values.  Since moving to 1.1.8 I didn't know if anything had changed in the PID algorithms and I wanted to re-run my autotunes again. It works just fine for E0 under 1.1.8, but it fails for E-1.

I saw the `contributing.md` page say that I needed to try Marlin 2.x first, but I have not done that as I am wary of trying out Nightly Build software marked as **Not for production use. Use with caution!** Sorry...

As a side note: Any thoughts on how stable 2.x is? Will it be going to a Stable branch soon?

### Steps to Reproduce

On an Anet A6 printer, compile and load 1.1.8 with `PIDTEMPBED` defined and `BED_LIMIT_SWITCHING` undefined.

Execute `M303 E-1 C8 S90`

**Expected behavior:** [What you expect to happen]

Get PID values for the heated bed.

**Actual behavior:** [What actually happens]
The system halts **immediately**! But, before it halts, it says `Error:Heating failed, system stopped! Heater_ID: bed`. I get the same result immediately no matter if the bed starts out unheated or pre-heated.

`Configuration.h` and `Configuration_adv.h` are attached:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2260786/Config.zip)

AnHardt, 2018-08-06:
If you want to stay with 1.1.8 stay with the error.
Maybe you missed the release of 1.1.9 where **this** bug **is** fixed.

JimF42, 2018-08-06:
**Doh!** I must of downloaded and prepped 1.1.8 with my auto-level changes just before 1.1.9 was released. 

Yeah, I will upgrade to 1.1.9 then.

Thanks.

closed 2018-08-06 by JimF42