MarlinFirmware/Marlin #11471: Trigorilla 2560 1.4 + tmc2208 , enstop are not working

Xenoyr, 2018-08-06:
Hi I open this thread because the endstop on my printer are not working, they are well connected but they arent relevate, so the steppers doesnt halt when they touch any endstop.. so I cannot do any homing..
this zip has the pinout schematics of the trigorilla board, and also my configuration of marlin.. 
[marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2261907/marlin.zip)
sorry I'm pretty noob.. and thanks for supporting

jscurtu, 2018-08-06:
I am using the same steppers and a trigorilla no issues for me.. see here:
https://github.com/jscurtu/Marlin_119_Anycubic_Kossel_Linear_Plus

jscurtu, 2018-08-06:
also check the logic of the endstops, they might be different if you are not using a delta:
```

//===========================================================================
//============================== Endstop Settings ===========================
//===========================================================================

// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
//#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG // a Z probe
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true  // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false  // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false  // set to true to invert the logic of the probe.
```

GMagician, 2018-08-06:
You may try M119 when endstop are released and when manually pressed

Xenoyr, 2018-08-08:
@jscurtu mine pinout of the board was completely different but anyway, in configuration adv I have those errors:

```
Configuration_adv.h:1022: error: missing binary operator before token "("

 #if HAS_DRIVER(TMC26X)

               ^

Configuration_adv.h:1250: error: missing binary operator before token "("

 #if HAS_DRIVER(L6470)

               ^

In file included from C:\Users\L\AppData\Local\Temp\arduino_build_516482\sketch\ultralcd.cpp:96:0:

C:\Users\L\AppData\Local\Temp\arduino_build_516482\sketch\ultralcd_impl_DOGM.h:46:20: fatal error: U8glib.h: No such file or directory

 #include <U8glib.h>

                    ^

compilation terminated.

exit status 1
missing binary operator before token "("

```

what should I do?

Futhermore, if I put my config adv, for any reason there is always error during compilation of arduino mega 2560 ext status 1, without any explanation

jscurtu, 2018-08-08:
It looks like you are missing the U8glib library. You need to search and add it in the adruino ide.

Xenoyr, 2018-08-08:
@jscurtu Installed, this time:
`In file included from C:\Users\L\AppData\Local\Temp\arduino_build_516482\sketch\MarlinConfig.h:31:0,

                 from C:\Users\L\AppData\Local\Temp\arduino_build_516482\sketch\G26_Mesh_Validation_Tool.cpp:27:

Configuration_adv.h:1022: error: missing binary operator before token "("

 #if HAS_DRIVER(TMC26X)

               ^

Configuration_adv.h:1250: error: missing binary operator before token "("

 #if HAS_DRIVER(L6470)

               ^

Uso la libreria U8glib alla versione 1.19.1 nella cartella: C:\Users\L\Documents\Arduino\libraries\U8glib 
exit status 1
missing binary operator before token "(" `

thinkyhead, 2018-09-01:
Assuming this is resolved.

closed 2018-09-01 by thinkyhead