MarlinFirmware/Marlin #11473: [1.1.9]

trendelkamp, 2018-08-06:
Hi Guys, 

with the latest Version 1.1.9 and also the bugfix Branch 1.1.x i do get the following error "**EEPROM datasize error**" while connecting to my printer. 
While investigating other similar problems i added #define DEBUG_EEPROM_READWRITE to my config file and afterwards i receive a little bit more explaining error "**Error:Field planner_leveling_active mismatch**".

So maybe you guys have some ideas because i did run out of them. 

BR

[Marinlin1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/2262236/Marinlin1.1.9.zip)

GMagician, 2018-08-06:
Try m502 and then m500

koelle25, 2018-08-21:
I had the same problem - your solution works fine for me @GMagician.
I just had to temporarily disable "disconnect on error" in octoprint to be able to input these commands.

closed 2018-08-21 by trendelkamp