MarlinFirmware/Marlin #11474: 1.1.9+ Will not fit on some controller boards

JimF42, 2018-08-06:
### Description

I guess this will just be more of an FYI for the developers/maintainers of Marlin, but 1.1.9 no longer fits on my controller board. It's an ANET board for my A6 printer. the Arduino compiler tells me that the compiled image will not be able to load on to the board. It turns out that I am short 920 bytes.

My rig has an added auto bed leveler that I have added to the configuration.h files, so perhaps an image without auto bed leveling will still fit.

I do understand that I don't actually have a "leading edge" controller here and that if I want to actually run 1.1.9 and probably 2.x, I will now need to start swapping out hardware.

I am more mentioning it since you may still have a call for 1.1.8 for a while because of this, or at least until these cheap controller boards start coming out with more memory.

marcio-ao, 2018-08-06:
@JimF42: I may be mistaken, but I think the ANET uses an ST7920 display. If so, you may want to try the "LIGHTWEIGHT_UI" option in "Configuration_adv.h" and see if it helps get you under the limit.

tobecwb, 2018-08-06:
I'm using a Arduino Mega 2560, with SD Card, Graphic Display, Bed Leveling with 3d touch sensor, TMC2130, and some other functions. All this option don't "fit" on Arduino Mega, even with `LIGHTWEIGHT_UI`. I have to disable the SD Card function (I'm using Octoprint, so, don't need anymore), or reduce the level bed probe points.

JimF42, 2018-08-07:
@tobecwb Interesting points... I was looking at other boards if I wanted to upgrade and I _assumed_ that the Mega would have had more PROGMEM (I have 124 KB on my Anet board). I guess I need to look closer when I get serious about that.

And yes, disabling the SD Card support now gives me just under 16K bytes left. I too am using Octoprint and don't need SD support.

thinkyhead, 2018-08-07:
Tim Hoogland (@houseofbugs) is doing a lot of work to reduce the binary size in his fork of Marlin. Try https://github.com/houseofbugs/TH3D-Marlin/releases and see if they help.

roel8032, 2018-08-08:
I have the same problem.
The software does not fit on the Anet A6 board. But when i switch back to:
#define LCD LANGUAGE en
in configuration.h it fits.
not with:
#define LCD LANGUAGE nl

roel8032, 2018-08-10:
It fits when you comment:
//#define EEPROM_CHITCHAT

Atatoth, 2018-08-10:
What for is chitchat then??

roel8032, 2018-08-10:
Don't know yet

closed 2018-08-14 by thinkyhead