MarlinFirmware/Marlin #11479: [1.1.9] Problem with Lost steps with TMC2208 and power loss feature

campy, 2018-08-07:
Hi, i configure new marlin 1.1.9, i use TMC2208 Standalone driver in MKS GenL to 24V

When i use 1.1.8 i can print to 15 mm/s Jerk and 3500mm/s acceleration without problems or missing steps, but in new marlin when i put this jerk and aceleration i have a lot of missing steps.

I configure driver to TMC2208 Standalone and try to TMC2208 whit library but same problem, always i get missing steps 

When i down jerk to 12mm/s and acceleration to 2000mm/s everything works ok but some times lost steps

Another curious thing is happen , when i define power loss features, my printer enter in a infinite bucle and reset when try to do G29 but i disable: #define S_CURVE_ACCELERATION and #define ADAPTIVE_STEP_SMOOTHING and now powerloss works perfectly

Anything have same problem with TMC2208?

Thanks and i apologize for my english

campy, 2018-08-07:
I try to modify this values for driver tmc2208 but same problem

#define MAXIMUM_STEPPER_RATE 400000
#define MINIMUM_STEPPER_PULSE 0
#define MINIMUM_STEPPER_DIR_DELAY 20

thinkyhead, 2018-09-01:
The meaning of jerk and acceleration might slightly differ. Keep adjusting until you find acceptable values that give good results. Don't forget `M502`, `M500` after any re-flash.

boelle, 2019-02-20:
@campy is this still an issue if you use bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead